realsenseai/librealsense #970: Up Board doesn't recognize D400 or ZR300 sensors...

labdlau, 2018-01-03:
| Required Info  |   |
|---|---|
| Camera Model | R200 / F200 / SR300 / ZR300 / D400 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |
| SDK Version | |

Running the latest Ubuntu 16.04 LTS on the original UP Board.  When running the realsense-viewer tool, I get a UVCVideo error that requires me to run:

sudo modprobe uvcvideo

but now I get the warning message:

03/01 13:10:35,500 WARNING [139849252468480] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0

while the viewer doesn't recognize the D400 sensor.

ev-mp, 2018-01-04:
Hello @labdlau, 

Librealsense requires specific permissions for USB devices. I would assume that you build from sources, therefore you need to apply udev rules according to this [ chapter](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#video4linux-backend-preparation) in the installation guide.

Please collect and post logs with the the output of `lsusb` and `dmesg -wT` when D400 is being connected to the Upboard.

Also note that while you don't mention the  librealsense version you're using, the two devices are mutually incompatible - ZR300 is supported by Librealsense1 (legacy branch) only, while for D400 requires librealsense2 to run.

dorodnic, 2018-01-19:
Hi @labdlau 
Do you still need assistance with this issue?

pribadihcr, 2018-01-27:
I also has similar problem

sudo realsense-viewer 
 27/01 20:07:28,718 WARNING [139911479253696] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0

RealSense-Customer-Engineering, 2018-02-27:
[Realsense Customer Engineering Team Comment] 
Hi @labdlau

Please make sure to connect D400 series camera into the USB3 OTG port of the UP board. My test configurations as follows have no this issue. "rs-capture" and "realsense-viewer" can work well.
1. Ubuntu 16.04 + kernel 4.4.0.104
2. D410 + FW 5.8.15
The D410 should be recognized as a SuperSpeed USB device.

Do you use D415 camera?

dorodnic, 2018-03-19:
Hi @labdlau @pribadihcr 
Feel free to open new issue if you need further assistance with installation

closed 2018-03-19 by dorodnic