realsenseai/librealsense #973: No pip install of pyrealsense2 for Linux

mkhansen-intel, 2018-01-04:
| Required Info  |   |
|---|---|
| Camera Model | R200 / F200 / SR300 / ZR300 / D400 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  |  4.4.0-103-generic |
| SDK Version | 2.9.0|

I tried to install the pyrealsense2 wrapper on Ubuntu using 'sudo pip install pyrealsense2' and it appears that it installs a windows only package. 

Is this expected or is there 'pip' installable Linux version too?

zivsha, 2018-01-04:
Hi @mkhansen-intel ,
The python PyPl package is still WIP on [my fork](https://github.com/zivsha/librealsense/pull/15).
The current version on [pypi/pyrealsense2](https://pypi.python.org/pypi/pyrealsense2) is indeed only for windows with python 2.7.
It would be great if you could test it on windows and let me know if it works for you.
Next week we will decide if we want to remove the current available package, or make it available for all supported platforms

mkhansen-intel, 2018-01-04:
Hi @zivsha, thanks for the fast response. Sorry I can't test it on Windows, but I'm happy to test it on Linux if you have a package version for that. I have previously built from source and used it on Linux, but was hoping for the pre-packaged version. Please let me know when that is ready, thanks.

mkhansen-intel, 2018-01-16:
@zivsha - any update on this?

zivsha, 2018-01-16:
Nothing yet, I'll be happy to update when we provide a cross platform package distribution. Meanwhile I've removed the package to prevent any confusion and undesired issue.

mkhansen-intel, 2018-01-17:
@zivsha - OK, thanks. I'll check back again in a week or two

mkhansen-intel, 2018-02-02:
@zivsha - Any update on when this might be merged?

zivsha, 2018-02-03:
I'm considering publishing a condo package istead of the PyPI distribution, considering this library is used by the scientific community.

dorodnic, 2018-07-22:
Seems like we are finally all set with `pipy`. 
If you encounter bugs with `pip install pyrealsense2` please open new issues.

closed 2018-07-22 by dorodnic