realsenseai/librealsense #975: Advanced Mode Parameters Explanation

xtrawurst, 2018-01-05:
| Required Info  |   |
|---|---|
| Camera Model | D400 | 
| Firmware Version | Any  | 
| Operating System & Version | Any |
| SDK Version | 2.x |

Is there a page explaining the different RealSense Advanced Mode parameters (such as "DS Neighbor Threshold" or "RSM Remove Threshold"), or at least their effects on the depth image?

Right now I am doing this manually by observation, modifying each parameter individually and noting down its effects on the depth. However that is very cumbersome and does not capture effects that parameters have _in combination_ by influencing each other.

dorodnic, 2018-01-06:
Hi @xtrawurst 
Unfortunately we do not have such documentation ready and available. 
You can however look into the various camera presets and how each changes these parameters.
In addition to hard-coded presets we are planning to publish a page with several additional presets in form of JSON files with pictures (in about a week).

xtrawurst, 2018-01-06:
Thanks. Would there be any concerns if I make the results of my experiments available to the public?

dorodnic, 2018-01-08:
Yes, you are welcomed to publish any observations you make

zenving, 2018-02-10:
Hi @xtrawurst Can you post your results of your experiments ? : )
Hi @dorodnic Have you post the explanation? : )

dorodnic, 2018-02-12:
Hi @zenving 
We have published [the page](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets) mentioned above that offers several presets (with their corresponding settings)
This can offer several examples of using advanced mode parameters to customise camera behaviour.

dorodnic, 2018-02-24:
Closing this issue, feel free to open a new one

closed 2018-02-24 by dorodnic

abertonica, 2019-11-08:
Is documentation for Advanced Mode parameters still lacking?

amusco-amz, 2020-11-11:
@rafaelspring did you ever end up publishing the results of your experiments? I would be very interested in reviewing any data you can provide.