realsenseai/librealsense #976: PointCloud2 output in RViz is frozen and does not update

Clare-Meng, 2018-01-05:
Camera Model: R200
Firmware Version: 1.0.72.10
Operating System & Version: Ubuntu 16.04
Kernel Version (Linux Only): 4.4.0.2-upboard

I am able to generate a depth and color image output, but the pointcloud2 output does not update in real time after less than a minute. It updates for a bit, but then the screen freezes and only the depth and color images continue to work. Does anyone know how to resolve this issue? Thanks.

icarpis, 2018-01-10:
Hello @Clare-Meng, could you please provide the output of ` ROS realsense_camera` topic?
In addition, please provide the published topics by typing `rostopic list` and try to read from the pointcloud2 topic by typing `rostopic echo /camera/depth/points` and `rostopic echo /camera/depth_registered/points`

RealSense-Customer-Engineering, 2018-02-27:
[Realsense Customer Engineering Team Comment] 
Hi @Clare-Meng

would like to know if you still need help with this issue?
thanks.

closed 2018-03-06 by RealSense-Customer-Engineering

djyang, 2018-03-13:
@RealSense-Customer-Engineering I have this exact same issue with the exact hardware/versions.

Camera Model: R200
Firmware Version: 1.0.72.10
Operating System & Version: Ubuntu 16.04
Kernel Version (Linux Only): 4.4.0.2-upboard

The pointcloud2 in rviz stops working after ~1min of running. It comes back if I restart the r200 nodelet, only to freeze again.

When reading the rostopic echo /camera/depth_registered/points, the data continues to output after the pointcloud2 freezes, but it eventually seems to get stuck (is there a delay in the data output?)

rostopic echo /camera/depth/points never stops.

Thanks, Daniel

chezhiane, 2018-07-03:
Kindly let me know the resolution. I face the same. Similar configurations.

cgpadwick, 2018-07-09:
Precisely the same issue is occurring for me.  The color and depth image windows update, but the pointcloud2 window stops updating after about 30 seconds.  When I close rviz and reopen it, the pointcloud2 windows does not display anything.  I have ctrl-C the window where I launched `roslaunch realsense_camera r200_nodelet_rgbd.launch` and relaunch it to get another 30 seconds of point cloud viewing.

`rostopic echo /camera/depth/points` and `rostopic echo /camera/depth_registered/points` both put out tons of information to the screen.