realsenseai/librealsense #979: python wrappers is consuming 100% of my CPU

Taprik, 2018-01-07:
| Required Info  |   |
|---|---|
| Camera Model | SR300  | 
| Firmware Version |  3.21.0.0 | 
| Operating System & Version | windows 10  |
| Kernel Version (Linux Only)  |   |
| SDK Version | 2.0|

I managed to compile a x64 binaries for python 3.6 x64. 
and then I use the dll provided with the sdk. The curious thing is that getting and displaying the color stream is consuming all my cpu.
When I use the intel viewer it needs 22% of the cpu. When I use the script opencv_viewer_example it needs most all the cpu.
Is there a way to have a descent cpu consumption with python.
By the way I need to have it works in x64 as I have other lib only working with python 3.6 x64 ...
I attach two screen capture, one with the intel_viewer
[image]
the other with python script (found in the examples folder)
[image]

zivsha, 2018-01-08:
Hi @Taprik , I marked this as a bug to further investigate the issue, and since I think we need the examples to provide a solution if required.
This is not a python issue, it is easily reproducible running the same code in C++ (or even the `rs-capture` example).
The viewer is not using the `pipeline`, and uses a number of message queues which is probably the cause of difference. After we investigate we will post the results.

Jiloc, 2018-01-16:
@zivsha Is it related to #744 ? Wouldn't it be good to release a version with OpenMP disabled?

zivsha, 2018-01-16:
Hi @Jiloc , we haven't got around to investigate this but OpenMP is a potential high CPU consumer. 
The python wrapper is available for self build and you can disable openMP via CMake.
I'll propose providing a version without OpenMP in the future.

Taprik, 2018-01-20:
I confirm that building with openMP to false solve the problem

Brekel, 2018-01-20:
I could reproduce this problem in C++ with the official 2.9.0 release but noticed 2 interesting things:

- 100% CPU usage when streaming depth + color but only minimal CPU usage when streaming only depth or depth + infrared
- recompiling from source with OpenMP disabled: minimal CPU usage for depth + color (recompile with OpenMP enabled 100% usage again)

Note that I installed 2.9.0 using the installer executable on Windows 10 and for source compilation I used the latest GitHub version (20th January)

zivsha, 2018-01-21:
@Brekel We use OpnMP to convert the color-stream's format from YUY to RGB, so your insights looks reasonable

RealSense-Customer-Engineering, 2018-02-28:
[Realsense Customer Engineering Team Comment] 
@Taprik   Can we close this issue now?

Taprik, 2018-02-28:
Yes you can.
Thanks a lot !

closed 2018-02-28 by RealSense-Customer-Engineering