realsenseai/librealsense #980: How to install python wrapper on Windows?

KeCh96, 2018-01-08:
The tutorial says: 

> When running cmake-gui, select the BUILD_PYTHON_BINDINGS option

But I cannot understand what exactly I should do? Do I need to install cmake first? Then run the cmake-gui?

I am confused because I have never used cmake...  Could you please tell me the concrete steps for installing python wrapper on Windows?

zivsha, 2018-01-08:
Hi @KeCh96 ,
The instructions there are a guide to building the python wrapper, not installing it.
If you want to install the wrapper you can download the **Intel.RealSense.SDK.exe** from [the release tab](https://github.com/IntelRealSense/librealsense/releases) and after installing, copy `pyrealsense2.pyd` and `realsense2.dll` from `C:\Program Files (x86)\Intel RealSense SDK 2.0\bin` next to your script.
If you want to build the wrapper from source (i.e - compile the code) then you need to follow the step in the python wrapper instructions (which I agree should we should rename from "Installation Guide").
In order to build from source, you first need to follow the [building from source guide for windows](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md) and then continue to build the python wrapper.

KeCh96, 2018-01-08:
Thank you very much!!

Unluckily, when I copied `pyrealsense2.pyd `and `realsense2.dll`  and import pyrealsense2, I got an error:

> ImportError: DLL load failed: The specified module could not be found

As you mentioned in the guide, this error may caused by python versions mismatch.  I use python 3, but the precompiled binaries shipped with the installer assume python2.7. Is there any solution for python3 wrapper?

zivsha, 2018-01-08:
Unfortunately no, the only way at the moment is by building it from source. We are planing to provide a PyPI package that will allow you to simply `pip install pyrealsense2` and it will have both python 2 and 3 support, but it is still work in progress...

Taprik, 2018-01-08:
Hy,
Here is how I compile the python bindings for python 3.6 x64
1. make sure you have the sdk installed
2. make sure you have visual studio with c++ components
3. open msbuild shell
4. cd to the repertory of the python wrappers
5. run cmake "." -G "your visual studio version"
6. open the vs project generated. You will have to add some librairies (librealsense...)
7. compile it. It will generate a pyrealsense2.pyd
8. drag this file in the correct folder regarding your installation with the dll mentioned in your post (for me it's in python36/Lib/site-packages)

It's working for me. Unfortunatly running a basic script with a small definition is eating all my cpu. Seems  to be a bug according to zivsha.

I will be interested in your results ;)

KeCh96, 2018-01-09:
It looks complicated to build python3 wrapper on Windows... What about building  python3 wrapper on Linux? Will it be easier? @zivsha @Taprik 

The tutorial says:

> Install Python and its development files via apt-get (Python 2 and 3 both work)
> sudo apt-get install python python-dev or sudo apt-get install python3 python3-dev
> **Note: The project will only use Python2 if it can't use Python3**

Does it mean that building python3 wrapper is also complicated on Linux?

zivsha, 2018-01-09:
The steps for Linux/Windows are pretty much identical.
Usually it is easier on Linux since common paths are used if you make install stuff they are easily found by CMake and the compiler (libs and headers). So try to build on Linux and let me know if we can assist

KeCh96, 2018-01-11:
My project can work on both Windows and Linux. According to my project, I only need a solution for any of the following 2 problems:
    1. Building python3 wrapper on Windows10
    2. Building python2 or python3 wrapper on Ubuntu16.04 

  I tried many methods, and my problems are as below: 

1. Building python3 wrapper on Windows10
     As you say, the only way to build python3 wrapper on Windows is to use CMake and build from source. I guess that building python2 wrapper on Linux may be easier, so I give up Windows solution.

2. Building python2 or python3 wrapper on Ubuntu16.04 
     I followed [the tutorial for Linux](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) . I run `realsense-viewer` and launched the SR300 GUI successfully. Then I run `sudo apt-get install librealsense2-dev` and `sudo apt-get install librealsense2-dbg` successfully.
    However, when I wanted to run `g++ -std=c++11 filename.cpp -lrealsense2`, I did not know where the "filename.cpp" is . So I skipped that step.
    Then I browsed [python wrapper tutorial for Linux](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python) and found I need to install CMake. Is there any method to build python wrapper without using CMake?
    I want to copy `pyrealsense2.pyd` and `realsense2.dll` to my program. But at this moment, I cannot find `pyrealsense2.pyd` and `realsense2.dll` in my system. Did I have some step missing?

    After these attempts, I found another tutorial [pyrealsense 2.2](https://pypi.python.org/pypi/pyrealsense/), and pray this can be an easy python wrapper building tool. But `pip install pyrealsense` run into an error:

> Collecting pyrealsense
>   Downloading pyrealsense-2.2.tar.gz (60kB)
>     100% |████████████████████████████████| 61kB 46kB/s 
> Requirement already satisfied: numpy in ./anaconda2/lib/python2.7/site-packages (from pyrealsense)
> Requirement already satisfied: cython in ./anaconda2/lib/python2.7/site-packages (from pyrealsense)
> Requirement already satisfied: pycparser in ./anaconda2/lib/python2.7/site-packages (from pyrealsense)
> Requirement already satisfied: six in ./anaconda2/lib/python2.7/site-packages (from pyrealsense)
> Building wheels for collected packages: pyrealsense
>   Running setup.py bdist_wheel for pyrealsense ... error
>   Complete output from command /root/anaconda2/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-k1ws2H/pyrealsense/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpQDkpd8pip-wheel- --python-tag cp27:
>   running bdist_wheel
>   running build
>   running build_py
>   creating build
>   creating build/lib.linux-x86_64-2.7
>   creating build/lib.linux-x86_64-2.7/pyrealsense
>   copying pyrealsense/extlib.py -> build/lib.linux-x86_64-2.7/pyrealsense
>   copying pyrealsense/offline.py -> build/lib.linux-x86_64-2.7/pyrealsense
>   copying pyrealsense/utils.py -> build/lib.linux-x86_64-2.7/pyrealsense
>   copying pyrealsense/extstruct.py -> build/lib.linux-x86_64-2.7/pyrealsense
>   copying pyrealsense/stream.py -> build/lib.linux-x86_64-2.7/pyrealsense
>   copying pyrealsense/core.py -> build/lib.linux-x86_64-2.7/pyrealsense
>   copying pyrealsense/__init__.py -> build/lib.linux-x86_64-2.7/pyrealsense
>   copying pyrealsense/constants.py -> build/lib.linux-x86_64-2.7/pyrealsense
>   running build_ext
>   building 'pyrealsense.rsutilwrapper' extension
>   creating build/temp.linux-x86_64-2.7
>   creating build/temp.linux-x86_64-2.7/pyrealsense
>   gcc -pthread -fno-strict-aliasing -g -O2 -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/root/anaconda2/lib/python2.7/site-packages/numpy/core/include -I/usr/local/include/librealsense -I/root/anaconda2/include/python2.7 -c pyrealsense/rsutilwrapper.cpp -o build/temp.linux-x86_64-2.7/pyrealsense/rsutilwrapper.o
>   cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
>   In file included from /root/anaconda2/lib/python2.7/site-packages/numpy/core/include/numpy/ndarraytypes.h:1777:0,
>                    from /root/anaconda2/lib/python2.7/site-packages/numpy/core/include/numpy/ndarrayobject.h:18,
>                    from /root/anaconda2/lib/python2.7/site-packages/numpy/core/include/numpy/arrayobject.h:4,
>                    from pyrealsense/rsutilwrapper.cpp:474:
>   /root/anaconda2/lib/python2.7/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
>    #warning "Using deprecated NumPy API, disable it by " \
>     ^
>   pyrealsense/rsutilwrapper.cpp:477:16: fatal error: rs.h: No file or directory
>   compilation terminated.
>   error: command 'gcc' failed with exit status 1
>   
>   ----------------------------------------
>   Failed building wheel for pyrealsense
>   Running setup.py clean for pyrealsense
> Failed to build pyrealsense
> Installing collected packages: pyrealsense
>   Running setup.py install for pyrealsense ... error
>     Complete output from command /root/anaconda2/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-k1ws2H/pyrealsense/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-taxuSu- [... cut ...]
> Command "/root/anaconda2/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-k1ws2H/pyrealsense/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-taxuSu-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-k1ws2H/pyrealsense/

I am almost crazy... I wonder the relationship between these tutorials. If I want to build python wrapper on Linux, which tutorial should I follow?

Could you please help me?

zivsha, 2018-01-11:
Hi @KeCh96 ,
I'll try to make some sense for you.
1. [pyrealsense 2.2](https://pypi.python.org/pypi/pyrealsense/2.2) is a wrapper for the previous version of the librealsense SDK ([legacy branch](https://github.com/IntelRealSense/librealsense/tree/legacy)), it is a [community supported wrapper](https://github.com/toinsson/pyrealsense), but again, for an older version (1.12.1). It [does not work with SDK version 2.x](https://github.com/toinsson/pyrealsense/issues/55).

2. As I've said there is no Linux package for the python wrapper we have for the SDK (talking about v 2.x now). This means that you cannot get the python wrapper with `sudo apt-get`. The [the tutorial for Linux](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) you followed installs the SDK on your system (it installs headers and shared-libraries `.so`) so that you may link your program to use the SDK, this is what the `g++ -std=c++11 filename.cpp -lrealsense2` step is - it is how to compile your program which has 1 file called `filename.cpp` which requires linkage to `librealsense2.so` (or `realsense2.dll` on windows). Which brings me to number 3...

3. Windows and Linux use different formats for libraries, so you cannot take `pyrealsense2.pyd` and `realsense2.dll` (which works on Windows) and use it in Linux (which needs `pyrealsense2.so` and `realsense2.so`)

4. "Building from source" means you need the source files (of the SDK), and you need to compile them into libraries. `CMake` is used to create a "buildable" file which you can use to compile the project. If you want to build anything from this SDK from source - you must use `CMake`, and then compile the output (either `make` in Linux or build with visual studio on Windows).
In order to build the python wrapper, you need to be able to build the SDK library (realsense2) with CMake, and if you are able to do that, you can then also add a flag to CMake to build the python wrapper (using the `-DBUILD_PYTHON_BINDINGS` flag), and then compile again. On Linux , you should be able to do the following (assuming you have all prerequisites mentioned in the [Linux guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) :

`sudo apt-get install cmake`
`cd ~`
`git clone https://github.com/IntelRealSense/librealsense.git`
`cd librealsense`
`mkdir build`
`cd build`
`cmake ../ -DBUILD_PYTHON_BINDINGS=true`
`make -j4`
After this step you'll have 3 files in the build folder starting with the name `librealsense2.so`
To find the python wrapper libraries:
`cd ./wrappers/python`
and you should find 3 files starting with the name `pyrealsense2.cpython-35m-x86_64-linux-gnu.so`

KeCh96, 2018-01-14:
Thank you very much!! @zivsha 

I followed your guide, but faced some error when `make -j4`:

> [ 59%] Linking CXX executable rs-pointcloud
In file included from /home/chengke/librealsense/wrappers/python/third_party/pybind11/include/pybind11/pytypes.h:12:0,
                 from /home/chengke/librealsense/wrappers/python/third_party/pybind11/include/pybind11/cast.h:13,
                 from /home/chengke/librealsense/wrappers/python/third_party/pybind11/include/pybind11/attr.h:13,
                 from /home/chengke/librealsense/wrappers/python/third_party/pybind11/include/pybind11/pybind11.h:36,
                 from /home/chengke/librealsense/wrappers/python/pybackend.cpp:1:
/home/chengke/librealsense/wrappers/python/third_party/pybind11/include/pybind11/common.h:70:20: fatal error: Python.h: No such file or directory
compilation terminated.
wrappers/python/CMakeFiles/pybackend2.dir/build.make:158: recipe for target 'wrappers/python/CMakeFiles/pybackend2.dir/pybackend.cpp.o' failed
make[2]: *** [wrappers/python/CMakeFiles/pybackend2.dir/pybackend.cpp.o] Error 1
CMakeFiles/Makefile2:1261: recipe for target 'wrappers/python/CMakeFiles/pybackend2.dir/all' failed
make[1]: *** [wrappers/python/CMakeFiles/pybackend2.dir/all] Error 2
make[1]: *** Waiting for an unfinished task....
/usr/bin/ld: /usr/local/lib/libglfw3.a(x11_window.c.o): undefined reference to symbol 'XConvertSelection'
//usr/lib/x86_64-linux-gnu/libX11.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
examples/capture/CMakeFiles/rs-capture.dir/build.make:98: recipe for target 'examples/capture/rs-capture' failed
make[2]: *** [examples/capture/rs-capture] Error 1
CMakeFiles/Makefile2:307: recipe for target 'examples/capture/CMakeFiles/rs-capture.dir/all' failed
make[1]: *** [examples/capture/CMakeFiles/rs-capture.dir/all] Error 2
/usr/bin/ld: /usr/local/lib/libglfw3.a(x11_window.c.o): undefined reference to symbol 'XConvertSelection'
//usr/lib/x86_64-linux-gnu/libX11.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
/usr/bin/ld: /usr/local/lib/libglfw3.a(x11_window.c.o): undefined reference to symbol 'XConvertSelection'
//usr/lib/x86_64-linux-gnu/libX11.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
examples/pointcloud/CMakeFiles/rs-pointcloud.dir/build.make:176: recipe for target 'examples/pointcloud/rs-pointcloud' failed
examples/multicam/CMakeFiles/rs-multicam.dir/build.make:98: recipe for target 'examples/multicam/rs-multicam' failed
make[2]: *** [examples/pointcloud/rs-pointcloud] Error 1
make[2]: *** [examples/multicam/rs-multicam] Error 1
CMakeFiles/Makefile2:417: recipe for target 'examples/multicam/CMakeFiles/rs-multicam.dir/all' failed
make[1]: *** [examples/multicam/CMakeFiles/rs-multicam.dir/all] Error 2
CMakeFiles/Makefile2:472: recipe for target 'examples/pointcloud/CMakeFiles/rs-pointcloud.dir/all' failed
make[1]: *** [examples/pointcloud/CMakeFiles/rs-pointcloud.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Then I use 'make -j4 -i' to ignore the errors, and get the following result:

> [ 20%] Built target realsense-file
In file included from /home/chengke/librealsense/wrappers/python/third_party/pybind11/include/pybind11/pytypes.h:12:0,
                 from /home/chengke/librealsense/wrappers/python/third_party/pybind11/include/pybind11/cast.h:13,
                 from /home/chengke/librealsense/wrappers/python/third_party/pybind11/include/pybind11/attr.h:13,
                 from /home/chengke/librealsense/wrappers/python/third_party/pybind11/include/pybind11/pybind11.h:36,
                 from /home/chengke/librealsense/wrappers/python/pybackend.cpp:1:
/home/chengke/librealsense/wrappers/python/third_party/pybind11/include/pybind11/common.h:70:20: fatal error: Python.h:  No such file or directory
compilation terminated.
[ 20%] Linking CXX shared library pybackend2.cpython-35m-x86_64-linux-gnu.so
c++: error: CMakeFiles/pybackend2.dir/pybackend.cpp.o:  No such file or directory
[ 29%] Built target pybackend2
[ 65%] Built target realsense2
[ 66%] Linking CXX executable rs-pointcloud
[ 66%] Linking CXX executable rs-multicam
[ 67%] Built target rs-save-to-disk
[ 68%] Linking CXX executable rs-capture
[ 69%] Linking CXX executable rs-align
/usr/bin/ld: /usr/local/lib/libglfw3.a(x11_window.c.o): undefined reference to symbol 'XConvertSelection'
//usr/lib/x86_64-linux-gnu/libX11.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
[ 69%] Built target rs-capture
/usr/bin/ld: /usr/local/lib/libglfw3.a(x11_window.c.o): undefined reference to symbol 'XConvertSelection'
//usr/lib/x86_64-linux-gnu/libX11.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
/usr/bin/ld: /usr/local/lib/libglfw3.a(x11_window.c.o): undefined reference to symbol 'XConvertSelection'
//usr/lib/x86_64-linux-gnu/libX11.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
[... 23 lines cut ...]
                 from /home/chengke/librealsense/wrappers/python/third_party/pybind11/include/pybind11/cast.h:13,
                 from /home/chengke/librealsense/wrappers/python/third_party/pybind11/include/pybind11/attr.h:13,
[... cut ...]

Then I went to `./wrappers/python`, and found `pyrealsense2.cpython-35m-x86_64-linux-gnu.so`, but its color is red, which means a compressed file. I run `cdll.LoadLibrary('pyrealsense2.cpython-35m-x86_64-linux-gnu.so')` in python IDE and got an error: 

> OSError: pybackend2.cpython-35m-x86_64-linux-gnu.so: cannot open shared object file: No such file or directory

I have written the path in `/etc/ld.so.conf` but the error still occur. I wonder whether the file is complied successfully or not?

zivsha, 2018-01-14:
> /home/chengke/librealsense/wrappers/python/third_party/pybind11/include/pybind11/common.h:70:20: fatal error: Python.h: No such file or directory

You are getting this because you did not follow step 2 in [Python wrapper guide](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python): 

2. Install Python **and its development files** via apt-get (Python 2 and 3 both work)
`sudo apt-get install python python-dev` or `sudo apt-get install python3 python3-dev`

KeCh96, 2018-01-16:
I solved the error `Python.h: No such file or directory` by adding include path, and run `make -j4` again. Now I can run `cdll.LoadLibrary('pyrealsense2.cpython-35m-x86_64-linux-gnu.so')` successfully. 

If I want to capture images with python, what should I do next? 

In the [Python wrapper guide](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python), you can just `import pyrealsense2 as rs` and capture images from SR300. But when I run `import pyrealsense2`, I got `ImportError: No module named pyrealsense2`.  I seem to need a file named `pyrealsense2.py`, but I cannot find it. I am confused about what should I do at this moment? @zivsha

zivsha, 2018-01-16:
You do not need `pyrealsense2.py`, you need `pyrealsense2` module (which is in `pyrealsense2.cpython-35m-x86_64-linux-gnu.so`). Try to run python from the folder in which `pyrealsense2...so` is located

KeCh96, 2018-01-19:
Thank you very much!! @zivsha 

I get error when I `import pyrealsense2.cpython-35m-x86_64-linux-gnu`, because the name contains '-'. After renaming this file, I build python2 wrapper on Ubuntu successfully!!

zivsha, 2018-01-21:
Hooray ❗️ 😃 

Hope this thread helps others, and that we finish work on python distribution soon.

closed 2018-01-21 by zivsha

amm385, 2019-11-21:
In case it helps future wanderers, I too ran into the `Python.h: No such file or directory` issue and had no luck installing python3-dev or even python3.5-dev. I followed [this tip](https://askubuntu.com/a/591935) and identified 4 lines of code that needed an update:

1. `librealsense/wrappers/python/third_party/pybind11/include/pybind11/detail/common.h:111-113`
2. `librealsense/wrappers/python/third_party/pybind11/include/pybind11/chrono.h:17`

For each of the 4 lines, I changed `#import <xxxx.h>`  to `#import <python3.5/xxxx.h>`

After this, `make` ran successfully. As mentioned above, the final step after `make install` was to find `pyrealsense2.cpython-36m-aarch64-linux-gnu.so` (in my case, in /usr/local/lib) and change its name to `pyrealsense2.so`.

FWIW I believe this is because I have both python2.7 and python3.5 installed but can't be sure

Jaffarali123, 2020-01-06:
> Hooray  
> 
> Hope this thread helps others, and that we finish work on python distribution soon.

Hello all, I am a newbie in working in python. I have installed Python 3.8 on windows 7 (64 bit). But i guess the code that is provided to me works on Linux. I am using PyCharm as IDE. I would be grateful if someone can help me in simulating one Linux file on my windows 7.

I am facing difficulty in understanding/running the following command.

Compile C++/Python wrappers: python setup.py build_ext --inplace