realsenseai/librealsense #981: How to send a frameset to a background thread

Jiloc, 2018-01-08:
| Required Info  |   |
|---|---|
| Camera Model | SR300 / D400 | 
| Firmware Version |   | 
| Operating System & Version | Windows 10  |
| Kernel Version (Linux Only)  |   |
| SDK Version | 2.9.0 |

Taking a look at the [API how to](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#do-processing-on-a-background-thread) if I want to process the frames in a background thread I should use a rs2::frame_queue. What if I need to send both the color and depth frames within a framset instead of a single frame?

dorodnic, 2018-01-08:
Hi @Jiloc 
`frameset` inherits from `frame` so you can just enqueue it to the frame queue. When getting the result, you will get a frame back and you can downcast it back to `frameset`: 
```cpp
frame f = q.wait_for_frame();
frameset fs = f.as<frameset>();
```

Jiloc, 2018-01-08:
I missed it, thank you!

closed 2018-01-08 by Jiloc