realsenseai/librealsense #983: RealSenseViewer crashes when Kinect v2 is connected

wldevries, 2018-01-08:
| Required Info  |   |
|---|---|
| Camera Model | R200 / F200 / SR300 / ZR300 / D400 | 
| Firmware Version |   | 

| Operating System & Version |   |
Windows 10 v 1709

| Kernel Version (Linux Only)  |   |

| SDK Version | |
SDK 2.0 version 2.8.2

Reproduce by connecting a Kinect (v2) to a computer and start the RealSense Viewer. It crashes immediately.

dorodnic, 2018-01-08:
Hi @wldevries 
That's good to know. I assume it has something to do with the fact that we try to connect to all UVC devices (searching for the integrated webcam if there is one). 
It would help if you have the call-stack of the crash.
We will also try to reproduce ourselves and resolve.

baSSiLL, 2018-03-02:
I want to confirm this bug is still there in the 2.10.0 release.

I've also built win32 debug from sources (691dc69228b0cbc0fc1718c7ecb69ab3b9611587) and got an exception with the following stack:
```
 	KernelBase.dll!751f08c2()	Unknown	Non-user code. Cannot find or open the PDB file.
 	[Frames below may be incorrect and/or missing, no symbols loaded for KernelBase.dll]		Annotated Frame
 	[External Code]		Annotated Frame
	realsense2.dll!librealsense::platform::wmf_uvc_device::get_pu_range(rs2_option opt) Line 692	C++	Symbols loaded.
 	realsense2.dll!librealsense::platform::retry_controls_work_around::get_pu_range(rs2_option opt) Line 523	C++	Symbols loaded.
 	realsense2.dll!librealsense::platform::multi_pins_uvc_device::get_pu_range(rs2_option opt) Line 774	C++	Symbols loaded.
 	realsense2.dll!librealsense::uvc_pu_option::get_range::__l2::<lambda>(librealsense::platform::uvc_device & dev) Line 43	C++	Symbols loaded.
 	realsense2.dll!librealsense::uvc_sensor::invoke_powered<librealsense::platform::control_range <lambda>(librealsense::platform::uvc_device &) >(librealsense::uvc_pu_option::get_range::__l2::librealsense::platform::control_range <lambda>(librealsense::platform::uvc_device &) action) Line 198	C++	Symbols loaded.
 	realsense2.dll!librealsense::uvc_pu_option::get_range() Line 46	C++	Symbols loaded.
 	realsense2.dll!librealsense::uvc_sensor::try_register_pu(rs2_option id) Line 680	C++	Symbols loaded.
 	realsense2.dll!librealsense::platform_camera::platform_camera(const std::shared_ptr<librealsense::context> & ctx, const std::vector<librealsense::platform::uvc_device_info,std::allocator<librealsense::platform::uvc_device_info> > & uvc_infos, const librealsense::platform::backend_device_group & group, bool register_device_notifications) Line 271	C++	Symbols loaded.
 	[External Code]		Annotated Frame
 	realsense2.dll!librealsense::platform_camera_info::create(std::shared_ptr<librealsense::context> ctx, bool register_device_notifications) Line 293	C++	Symbols loaded.
 	realsense2.dll!librealsense::device_info::create_device(bool register_device_notifications) Line 51	C++	Symbols loaded.
 	realsense2.dll!rs2_create_device(const rs2_device_list * info_list, int index, rs2_error * * error) Line 316	C++	Symbols loaded.
 	[External Code]		Annotated Frame
````
The reason appears to be that **wmf_uvc_device._video_proc** field is `NULL`.

dorodnic, 2018-03-19:
Hi @baSSiLL 
Thank you for detailed call-stack and crash info. We will add some extra checks around this area of code, but we still only have just Kinect 1 at our lab.

baSSiLL, 2018-03-19:
Hi, @dorodnic 
Let me know when this update is published to GitHub, I'll test whether it fixes the issue.

dorodnic, 2018-04-02:
Hi @baSSiLL 
Did you get a chance to try the new release (v2.10.2)?

baSSiLL, 2018-04-02:
Yes @dorodnic , I did - and it works! No more crashes while Kinect 2 is connected. Actually I tested on the next day after release, but forgot to get back to you with results.
Thanks for quick fix guys.

dorodnic, 2018-04-03:
Thanks for the feedback.

closed 2018-04-03 by dorodnic