realsenseai/librealsense #986: rs2::frame lifetime

sogartar, 2018-01-09:
| Required Info  |   |
|---|---|
| Camera Model |  | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |
| SDK Version | 2.8.3 |

What is the lifetime of rs2::frame?
I have a suspicion that it is alive as long as the sensor is open. I started getting some null refs when I reordered my deinitialization.
If my suspicion is true, then the frame needs to have weak ptr mechanics and be possible to promote it to strong ptr, so it does not spontaneously die while you are still working with it.

dorodnic, 2018-01-09:
`frame` can outlive the `sensor` it came from. There is some minor corner cases (for example, if you use depth frame to query distance and the depth sensor is no longer present it can create issues, but in general, you can delete the sensor, device and rest and still hold frames.

dorodnic, 2018-01-09:
If you share your code we can help troubleshoot it. The expectation is that all SDK objects must be able to delete in any order.

sogartar, 2018-01-10:
@dorodnic , thanks for the clarification.
I was not in the aforementioned corner cases. The problem was with my code.

closed 2018-01-18 by dorodnic