realsenseai/librealsense #989: error in copy_dependencies.bat in wrapper/csharp

bastiankayser, 2018-01-09:
| Required Info  |   |
|---|---|
| Camera Model | waiting for D400 | 
| Firmware Version |   | 
| Operating System & Version | Windows 1709 Build 16299.192  |
| Kernel Version (Linux Only)  |   |
| SDK Version | 2.9.0.0 |

Error in Build of wrapper/charp/Intel.RealSense.csproj Pre-Build event:
The system cannot find the batch label specified - nolocal
Pre-Build event is:
$(SolutionDir)\copy_dependencies.bat $(PlatformName) $(ConfigurationName) "$(SolutionDir)" "$(TargetDir)"
the batch file copy_dependencies must be changed to:
echo Configuration = %2
echo Platform = %1

if not exist "%3..\..\build\%2" goto nolocal
echo Copying from local compilation
xcopy /Y "%3..\..\build\%2\realsense2.dll" %4
exit 0
:nolocal
if not exist "C:\Program Files (x86)\Intel RealSense SDK 2.0" goto noinstall
echo Copying from installed SDK
xcopy /Y "C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\%1\realsense2.dll" %4
exit 0
:noinstall
echo "realsense2.dll NOT FOUND! Please build it from source using CMake or install Intel RealSense SDK 2.0 from GitHub releases"
exit 1

dorodnic, 2018-01-09:
Hi @bastiankayser 
You are correct. I assume it only worked with `build` folder existing. Would you like to submit a pull-request?

bastiankayser, 2018-01-09:
Yes will do.

closed 2018-01-13 by dorodnic