realsenseai/librealsense #993: RealSense Viewer with D415 and D435 - actual stereo module resolution only half that of set resolution

spigge, 2018-01-10:
I'm testing the Realsense D415 and D435 demo cameras (Firmware version 05.08.15.00)  with RealSense Viewer v2.9.0 on Windows 10 latest update. No matter which depth resolution I set on the stereo module, I only get half that on screen. For instance, if I set resolution to 640 x 480, I get 320 x 240. Is this a bug or is there another reason for this limitation? Thanks.

dorodnic, 2018-01-10:
Hi @spigge 
Please disable post-processing:
[image]
In particular you can disable the `decimate` filter, that is intended to reduce the resolution of the depth image preserving only pixels with good depth.
These filters are not enabled by default when you write applications, but we do enable them in the viewer.

spigge, 2018-01-10:
Thanks a lot for that, I do get full resolution if I deactivate the decimate filter. It makes sense of course in retrorespect.

closed 2018-01-10 by spigge

dorodnic, 2018-01-10:
This is good feedback. 
We knowingly enabled post-processing by default in the last release (2.9.0). If you compare in the 3D tab post-processing will offer less noisy point-cloud that otherwise looks the same. 
Perhaps we need to offer some visual indication for the fact that the stream is augmented.

spigge, 2018-01-10:
That makes sense to have post processing on by default since, in terms or spatial resolution, I cannot really distinguish between 1280 x 720 with decimation filter off and 640 x 360 with decimation filter on. Depth noise is less with the filter on which is an advantage. If I instead compare the spatial resolution between a set resolution of 640 x 360 without decimation and 640 x 360 with decimation (i.e., set resolution of 1280 x 720), the latter case has better spatial resolution. So it is perhaps a little unfair to nominally downgrade the resolution when activating the decimation filter since it does not seem to be the same thing as actually reducing the spatial resolution.

spigge, 2018-01-10:
While I have you in this thread. We want to integrate the D400 into our c++ windows store app. Any idea if this will work? If you think this is a topic of general interest, we can make a new thread I guess.

dorodnic, 2018-01-10:
You are correct to state that Windows store environment is limiting. Out of the box it will not work. 
Intel is working with Microsoft to enable SR300 and D400 cameras using Microsoft's new API for 3D-cameras. Once this work is done and public, we can wrap it in librealsense as another backend to simplify porting applications between platforms, but that would take time. 
I suggest raising this issue on [Intel communities](https://communities.intel.com/community/tech/realsense), since we are not directly in control of this activity. You can of course raise it here to gather more feedback from people in the community and SDK engineers.