realsenseai/librealsense #995: librealsense-2.9.0/src/device_hub.cpp:83: sequence point error ?

dcb314, 2018-01-10:
librealsense-2.9.0/src/device_hub.cpp:83:27: warning: operation on '((librealsense::device_hub*)this)->librealsense::device_hub::_camera_index' may be undefined [-Wsequence-point]

Source code is

  _camera_index = ++_camera_index % _device_list.size();

Better code would be

  _camera_index = (1 +_camera_index) % _device_list.size();

zivsha, 2018-01-10:
Hi @dcb314 , care to open a pull request with the change?

dcb314, 2018-01-10:
No thanks, I just find them, I don't fix them.

It might be worth adding flag -Wsequence-point (or maybe -Wall) to gcc build make sure this
kind of simple error doesn't occur again in future.

dorodnic, 2018-01-14:
Thank you, @dcb314

closed 2018-01-14 by dorodnic