realsenseai/librealsense #999: Depth resolution at 0.2 meter

mandelyoni, 2018-01-11:
| Required Info  |   |
|---|---|
| D435 | 

What is the depth resolution at a distance of 0.2 meter?

Thanks

jlgarcia75, 2018-01-23:
Hello @mandelyoni,

The resolution of the D400 cameras is less than 1% of the distance from the camera. This means that if your distance is 0.2m, your resolution will be somewhere between 1mm to 2mm.

Regards,
Jesus G.
Intel Customer Support

agrunnet, 2018-02-17:
The depth resolution scales with resolution mode and with distance and with camera. 
I will here talk about the RMS error (the depth noise of a plane). I will also assume you have a well textured plane you are looking at. (The projector clearly works to create texture as well, but you do get better results passively IF there is a texture already).

So for D435, in 848x480, the minimum operating distance (minZ) is 17cm. At 20cm, the RMS error should theoretically be <0.3mm. So will you see this?  You can, but you need to make sure you have a flat surface of course. Then you need to change the depth units. By default the camera reports depth in 1000um steps, or 1mm.  You can change this under the Advanced Mode to 100um for example. Finally, there is the issue that we currently focus the lenses with hyperfocal at 50cm. Which means images will be a bit blurred at 20cm, so the actual depth degrade a bit. 

For the D415, at 1280x720, the minZ is 44cm. So you you to see anything at 20cm you have two options. You can reduce the resolution to 640x360 and your minZ is 22cm. Or go to 320x240 and the minZ is 11cm. At 20cm the RMS error will be ~200um. Again, there will be the same caveats as stated before. 

For a stereo system the RMS error will scale quadratically. Another way of saying this is that the % error (which is % of distance) will scale linearly. So if you have 1% error at 4m, then you will have 2% at 8m, and 4% at 16m. And conversely you will have 0.5% at 2m, and 0.25% at 1m.   
    
One last thing to think about: while I have mentioned measuring a plane at 20cm away, since we are ranging based on stereo vision, we will start having more serious occlusion issues when a real object (not a plane) gets closer. Think of a finger approaching your eyes. At some point your left eye will only see the right side of the finger and the right eye will only see the left side, so there will not be any correspondance matching between left and right images. And you get a real head ache. The ASIC does too. 

I hope this helps.

jlgarcia75, 2018-02-28:
Hello @mandelyoni,

Do you still need help with this issue?

Best regards,
Jesus G.
Intel Customer Support

closed 2018-03-02 by jlgarcia75