realsenseai/librealsense #1006: RAW8 is not natively supported on Ubuntu 17.10 Artful with patched kernel 4.13.0-25-generic

compiaffe, 2018-01-12:
| Required Info  |   |
|---|---|
| Camera Model | D430 | 
| Firmware Version |  05.07.05.00 5.38.0.0 | 
| Operating System & Version | Ubuntu 17.10  |
| Kernel Version (Linux Only)  | 4.13.0-25-generic  |
| SDK Version | v2.9.0 |
Intel RealSense 430 with Tracking Module

When trying to stream the fisheye camera in the realsense-viewer, I get a pop up error message complaining about:
```
Backend in rs2_open_multiple(sensor:0x7f29380018d0, profiles:0x7f293823c650, count:1):
The requested pixel format 'RAW8' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Inappropriate ioctl for device
```

The terminal output of realsense-viewer shows the same:
```
realsense-viewer 
 12/01 14:11:05,642 WARNING [140526789818112] (sensor.cpp:265) Unregistered Media formats : [ UYVY ]; Supported: [ ]
 12/01 14:11:05,645 WARNING [140526789818112] (backend-v4l2.cpp:1100) Pixel format 38574152-1a66-a242-9065-d01814a likely requires patch for fourcc code RAW8!
 12/01 14:11:10,673 WARNING [140526789818112] (types.cpp:57) set_pu(id=9) failed! Last Error: Input/output error
 12/01 14:11:10,674 WARNING [140526789818112] (sensor.cpp:626) Exception was thrown when inspecting properties of a sensor
 12/01 14:11:10,674 ERROR [140526789818112] (types.h:152) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument
 12/01 14:11:10,674 WARNING [140526789818112] (sensor.cpp:626) Exception was thrown when inspecting properties of a sensor
 12/01 14:11:12,454 WARNING [140526657447680] (backend-v4l2.cpp:809) Empty frame has arrived.
 12/01 14:11:12,454 WARNING [140526657447680] (backend-v4l2.cpp:809) Empty frame has arrived.
 12/01 14:11:12,454 WARNING [140526657447680] (backend-v4l2.cpp:809) Empty frame has arrived.
 12/01 14:11:12,454 WARNING [140526657447680] (backend-v4l2.cpp:809) Empty frame has arrived.
 12/01 14:11:14,038 WARNING [140527111000128] (archive.cpp:305) The user was holding on to 1 frames after stream 0x0x173e5b0 stopped
 12/01 14:11:14,047 WARNING [140526798210816] (archive.cpp:315) All frames from stream 0x0x173e5b0 are now released by the user
 12/01 14:11:14,057 WARNING [140527111000128] (archive.cpp:305) The user was holding on to 1 frames after stream 0x0x173e5b0 stopped
 12/01 14:11:14,057 WARNING [140527111000128] (archive.cpp:305) The user was holding on to 1 frames after stream 0x0x173e5b0 stopped
 12/01 14:11:14,057 WARNING [140527111000128] (archive.cpp:315) All frames from stream 0x0x173e5b0 are now released by the user
```

and the output of dmesg when plugging in the camera complains about an unknown video format:
```
[   62.660107] usb 2-1: new SuperSpeed USB device number 3 using xhci_hcd
[   62.680894] usb 2-1: New USB device found, idVendor=8086, idProduct=0ad5
[   62.680896] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   62.680897] usb 2-1: Product: Intel(R) RealSense(TM) 430
[   62.680898] usb 2-1: Manufacturer: Intel(R) RealSense(TM) 430
[   62.680899] usb 2-1: SerialNumber: 012345678901
[   62.681668] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[   62.681672] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[   62.681830] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 430 (8086:0ad5)
[   62.683346] uvcvideo: Unable to create debugfs 2-3 directory.
[   62.683418] uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) 435 RGB was not initialized!
[   62.683419] uvcvideo 2-1:1.0: Entity type for entity Processing 2 was not initialized!
[   62.683420] uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) 435 RGB was not initialized!
[   62.683421] uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
[   62.683523] input: Intel(R) RealSense(TM) 430 as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/input/input25
[   62.683878] uvcvideo: Unknown video format 38574152-1a66-a242-9065-d01814a8ef8a
[   62.683881] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 430 (8086:0ad5)
[   62.684609] uvcvideo: Unable to create debugfs 2-3 directory.
[   62.684695] uvcvideo 2-1:1.3: Entity type for entity Processing 11 was not initialized!
[   62.684697] uvcvideo 2-1:1.3: Entity type for entity Extension 12 was not initialized!
[   62.684697] uvcvideo 2-1:1.3: Entity type for entity Camera 10 was not initialized!
```

---
The kernel is 4.13.0-25-generic and patched using the 
```librealsense/scripts/patch-realsense-ubuntu-xenial.sh``` following the [instructions here](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md#video4linux-backend-preparation).

Could this be an issue with running the patch for Ubuntu Xenial on Ubuntu Artful (17.10)?

dorodnic, 2018-01-12:
Hi @compiaffe 
Yes, `RAW8` is one of the "problematic" formats not yet upstreamed as far as I remember. 
Most likely this is an issue with the kernel patch (either the patch is not compatible with 17.10 or there was some other problem during installation). We will be happy to help troubleshoot with you.

compiaffe, 2018-01-15:
Hi @dorodnic 
thank you for the quick answer. I reran `patch-realsense-ubuntu-xenial.sh` - it seems as if the script does exit quietly just after the installing the dependencies:
```
sudo ./scripts/patch-realsense-ubuntu-xenial.sh 

Remove all RealSense cameras attached. Hit any key when ready

Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.4ubuntu1).
git is already the newest version (1:2.14.1-1ubuntu4).
linux-headers-generic is already the newest version (4.13.0.25.26).
0 to upgrade, 0 to newly install, 0 to remove and 5 not to upgrade.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
```

It seems that the `choose_kernel_branch` function defined in patch-utils.sh exits with error_code 1 without notifying the user...
Which kernel branch should be used for my 4.13.0 kernel?

ev-mp, 2018-01-15:
Hello @compiaffe ,
The patches we provide are for Ubuntu 16 with LTS kernels 4.4 and 4.10.
We haven't ported them to Ubuntu 17 with kernel 4.13 yet, so we add this request to the back log now.

In the meantime the fastest route to enable FishEye streaming would be to downgrade the kernel to 4.10 and then to apply the patch.

compiaffe, 2018-01-15:
Hi @ev-mp again thank you for the very quick answer.
I downgraded to kernel to:
```
$ uname -r
4.10.17-041017-generic
```
and the script runs quite a bit further, but then fails:
```
 $ sudo ./librealsense/scripts/patch-realsense-ubuntu-xenial.sh 

Remove all RealSense cameras attached. Hit any key when ready

Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.4ubuntu1).
git is already the newest version (1:2.14.1-1ubuntu4).
linux-headers-generic is already the newest version (4.13.0.25.26).
The following packages were automatically installed and are no longer required:
  libkeybinder0 libvte-common libvte9 python-gconf python-keybinder python-vte
Use 'sudo apt autoremove' to remove them.
0 to upgrade, 0 to newly install, 0 to remove and 5 not to upgrade.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
The kernel has modified files:
	modified:   drivers/media/usb/uvc/uvc_driver.c
	modified:   drivers/media/usb/uvc/uvcvideo.h
Proceeding will reset all local kernel changes. Press 'n' within 10 seconds to abort the operation
Do you want to proceed? [Y/n]y
Update the folder content with the latest from mainline branch
remote: Total 0 (delta 0), reused 0 (delta 0)
From git://kernel.ubuntu.com/ubuntu/ubuntu-xenial
 * branch                hwe        -> FETCH_HEAD
Resetting local changes in ubuntu-xenial-hwe folder
 HEAD is now at e2fefd10d UBUNTU: Ubuntu-hwe-4.13.0-29.32~16.04.1
Applying realsense-uvc patch
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 203 with fuzz 2 (offset 15 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 127 with fuzz 2 (offset -16 lines).
patching file drivers/media/v4l2-core/v4l2-ioctl.c
Hunk #1 FAILED at 1217.
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/v4l2-core/v4l2-ioctl.c.rej
patching file include/uapi/linux/videodev2.h
Hunk #1 FAILED at 661.
1 out of 1 hunk FAILED -- saving rejects to file include/uapi/linux/videodev2.h.rej
```

compiaffe, 2018-01-15:
It seems as if the script will downloaded the 4.13 again...
```
 HEAD is now at e2fefd10d UBUNTU: Ubuntu-hwe-4.13.0-29.32~16.04.1
```

ev-mp, 2018-01-15:
Hi, at the beginning of January Ubuntu updated their github branches names so the branch 410 turned out to be 4.13.
We adjusted the patches to the new convention last week and pushed it into the `development` branch.
Please get this change or cherry pick [this commit](https://github.com/IntelRealSense/librealsense/commit/5ab9cfc70cc50afa573e1dcd3e6bba5a5b08b393).
After that please manually delete the folder with Ubuntu 4.13 sources (ubuntu-xenial-hwe) and reapply the patch.

ev-mp, 2018-01-15:
Hello @compiaffe ,
There is one more thing - the kernel you reported is 4.10.17-041017-generic.
Which is not Ubuntu kernel that follows 4.10.**0-xx** pattern (4.10.0-42 currently).
Please follow the instructions from Ubuntu [official site](https://wiki.ubuntu.com/Kernel/LTSEnablementStack)

compiaffe, 2018-01-15:
In retrospect I don't think this hassle is worth running 17.10 for - so I will downgrade the OS to 16.04.3 LTS. But this is what I have managed to do so far:

I tried to install a 4.10.0-42 kernel, but the one I found wouldn't boot ([image](https://packages.ubuntu.com/xenial/amd64/kernel/linux-image-4.10.0-42-generic), [header-generic](https://packages.ubuntu.com/xenial/amd64/linux-headers-4.10.0-42-generic), [header_all](https://packages.ubuntu.com/xenial/all/linux-headers-4.10.0-42/download). 

Therefore, 
- I installed `4.10.0-041000-generic` from http://kernel.ubuntu.com/~kernel-ppa/mainline/v4.10/
- `git cherry-pick  -m 1 5ab9cfc70cc50afa573e1dcd3e6bba5a5b08b393` 
-  undid the changes to `scripts/patch-realsense-ubuntu-xenial.sh` introduced by the cherry-pick to get the script running
- selected the standard answers of the interactive configuration routine (partial log here: https://gist.github.com/compiaffe/50943043746274420ea06ec8be2a941a)

2 Modules were build successfully:
```
Patched kernels modules were created successfully

Handle videodev :
	Module videodev  is in use by uvcvideo
	Unloading dependency uvcvideo
	Module is resident, unloading ...  succeeded. 
Applying the patched module ...  succeeded
	Reloading dependency uvcvideo ...  succeeded. 

Handle uvcvideo :
Module is resident, unloading ...  succeeded. 
Applying the patched module ...  succeeded
	Reloading dependency uvcvideo ...  succeeded. 

Handle hid-sensor-accel-3d :
Applying the patched module ... modprobe: ERROR: could not insert 'hid_sensor_accel_3d': Unknown symbol in module, or unknown parameter (see dmesg)
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  hid-sensor-accel-3d  module was reloaded
```

Now when I start realsense-viewer I can select the FOV module and stream. I still cannot stream the Motion Module.

Thank you for all the help.

closed 2018-01-15 by compiaffe

dorodnic, 2018-01-15:
If you fetch the latest code from the development branch you should be able to start the Motion Module.

compiaffe, 2018-01-18:
Back on a fresh installation of 16.04 I tried again:
I still have issues running the 4.10.0-42 kernel, so I went with the 4.10.0-28 kernel instead.

- I installed the librealsense at the HEAD of the `development` branch. 
- Ran the .../scripts/patch-realsense-ubuntu-xenial.sh 
- I did __not__ cherry-pick anything or reverted any changes as I did on 17.10 as described above.

The patch-realsense... script usually queries me to unplug any RealSense devices. It didn't this time (there seems to be something off here: https://github.com/IntelRealSense/librealsense/blob/development/scripts/patch-realsense-ubuntu-xenial.sh#L7). However, just pressing enter progressed as before. All modules build and `realsense-viewer` displays the fisheye and depth stream. However, I still cannot access the IMU data and rs-capture and rs-multicam still fail.

dorodnic, 2018-01-18:
Most likely the camera is an early engineering sample, based on the firmware version. I suspect this is the reason for IMU failure.

compiaffe, 2018-01-19:
Yes, that is correct. I am about to get the release D435 - will test with that one again.

compiaffe, 2018-02-05:
Just to finish this thread:
My new D435 and D415 do not have any IMU modules any more, so that issue is gone.
These cameras also work fine on a laptop, but are still a bit tricky for use on a Jetson as per #1095 
Thanks for all the help @ev-mp and @dorodnic

mjscosta, 2018-02-26:
Hi, it was not clear for me, the only kernel that will work, is the 4.10, right, or you're supporting also the 4.13 (there as mention about branches changing name) ?

https://wiki.ubuntu.com/Kernel/Support