realsenseai/librealsense #1011: Accessing  RS options in pyrealsense2

bitzombie1, 2018-01-15:
| Required Info  |   |
|---|---|
| Camera Model | SR300  | 
| Firmware Version | 3.21  | 
| Operating System & Version | Ubuntu 14.04 |
| Kernel Version (Linux Only)  | 4.12.14 generic  |
| SDK Version | 2.0|

I am able to acquire color and depth streams from my RS300 using code similar to the opencv_viewer_example.py but I noticed the depth values max out at around 4 ft (1.2 m) by default. I am able to extend the depth range using the rs-sensor-control example and adjusting the “motion vs range” option. In my previous C++ projects I was able to get_option and set_options on the device object and I would like to be able to do that now in Python but can’t for the life of me find out how to access rs_options through the rs.pipeline() or rs.config() objects.  Can you direct me to the proper API or possibly a code example of how I can change the options?  Thanks

dorodnic, 2018-01-15:
Hi @bitzombie1 
The quick answer is something like:
```cpp
profile = pipeline.start(config)
depth_sensor = profile.get_device().first_depth_sensor()
depth_sensor.set_option(...);
```
@zivsha can correct me

zivsha, 2018-01-15:
Hi @bitzombie1 ,

The python API is almost 1-1 as the C++ API. Just like the `rs-sensor-control` example, you can use python API to get/set options via a sensor instance.
The `pipeline` encapsulates the device (and its sensors) so there is not API for controlling options via the `pipeline` itself. 
The way to do that is using the `pipeline_profile` from which you can `get_device()` and then you are back to the device and sensor APIs.
The `pipeline_profile` is the return value of `pipeline.start()` or `config.resolve()`.
The `align-depth2color.py` example show something similar to that...

Here is a snippet for getting what you want:
```py
import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
profile = config.resolve(pipeline) # does not start streaming
profile = pipeline.start(config) # Start streaming
depth_sensor = profile.get_device().first_depth_sensor()
exposure = depth_sensor.get_option(rs.option.exposure)
exposure = depth_sensor.set_option(rs.option.exposure, 0)
```

bitzombie1, 2018-01-15:
Thanks for the quick response! So when I try that code snippet I get:

File "df2.py", line 36, in <module>
    motion_range = depth_sensor.get_option(rs.option.motion_range)
TypeError: get_option(): incompatible function arguments. The following argument types are supported:
    1. (self: rs2::options, option: pyrealsense2.option) -> float

Invoked with: <pyrealsense2.depth_sensor object at 0x7f14ca253f60>, option.motion_range

I got a similar error when trying to get  rs.option.exposure

incidentally when I try to run align_depth2color.py I get:
module 'pyrealsense2' has no attribute 'align' 

It’s making me wonder if I messed up the installation somewhere but the opencv_viewer_example.py
works fine.

zivsha, 2018-01-16:
The first issue looks like you don't have the alias `rs` for `pyrealsense2` (possibly you use it for something else?), test if `motion_range = depth_sensor.get_option(pyrealsense2.option.motion_range)` works for you.

Regarding the `align` issue, this indeed looks like a version issue, align was introduced in 2.8.3

zivsha, 2018-01-24:
@bitzombie1 any updates?

bitzombie1, 2018-01-27:
I have been struggling with compiling the new version 2.9.1 but it's a new day,new build and I should be able to actually try your suggestions today.

bitzombie1, 2018-01-27:
So I was successful in installing librealsense 2.9.1 Cmake with python wrappers , make completed without error, make install worked fine, and since I’m using a virtual environment I copied  pybackend2.* and pyrealsense2.* from ~/librealsense/build/wrappers/python into my virtual environment. When I run the example align-depth2color.py I get the following traceback. I’m not a full time Linux programmer, I can usually trace down errors, but I’m not sure where to start on this one. If you have any idea why this would fail in this way or could point me in the right direction I would be thankful. 

(cv) charles@UP-16-fd:~/librealsense/wrappers/python/examples$ python align-depth2color.py 
Traceback (most recent call last):
  File "align-depth2color.py", line 25, in <module>
    profile = pipeline.start(config)
RuntimeError: Failed to resolve request. No device found that satisfies all requirements
*** Error in `python': free(): invalid pointer: 0x00007f412d988c8c ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f412eaec7e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f412eaf537a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f412eaf953c]
/home/charles/.virtualenvs/cv/lib/python3.5/site-packages/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(+0x1676e7)[0x7f412d96c6e7]
/home/charles/.virtualenvs/cv/lib/python3.5/site-packages/pyrealsense2.cpython-35m-x86_64-linux-gnu.so(+0x17ec42)[0x7f412d983c42]

zivsha, 2018-01-28:
> RuntimeError: Failed to resolve request. No device found that satisfies all requirements
Is an exception throw from `pipeline.start`, telling you that the configuration you requested from the pipeline could not be resolved.
In the  align-depth2color.py script the configuration tries to enable the following streams:
```py
config.enable_stream(rs.stream.depth, 640, 360, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
```
First, try to remove these 2 lines and see if you are able to run the example (in this case the default configuration will be used). If so, the camera you use does not support these streams for some reason (which camera are you using? there might be another issue here).
If it doesn't work - please paste the error here, and also open the RealSense Viewer and try to stream using it, to make a sanity check that your device actually works and that the librealsense version you use is valid.

bitzombie1, 2018-01-28:
So just to make sure I reinstalled librealsense (build 2.9.1) zip from the release page instead of cloning it. Got the same error  running align-depth2color.py . I’m testing with both  the Intel RS300 and the Creative Blaster X rebrand (essentially the RS300 in a new box) . Both have updated firmware and both work fine using the Intel DCM viewer. I tried removing the two lines you mentioned and indeed it worked but exiting out of the program I get an identical error  as @zhutq found in issue #1087 so I’m following that thread for the moment .

zivsha, 2018-01-29:
You can always run the [RealSense Viewer](https://github.com/IntelRealSense/librealsense/releases/download/v2.9.1/Intel.RealSense.Viewer.exe) to verify that the camera is fine and that it supports the stream you want.

I think the example code should have been:
```py
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 360, rs.format.bgr8, 30)
```
(Switch resolutions for depth and color).
It would be helpful if you could update issue 1087 with what you are seeing so that we have more details (as I failed to reproduce it the first time)

bitzombie1, 2018-02-03:
As far as the original question that spawned this thread I am now able to get and set RS options using the examples in your first reply.  You can close this thread. Thank you very much.

closed 2018-02-03 by bitzombie1

OctaM, 2018-05-10:
Is there any way to access the color sensor? I can't figure out a way to do that because I'm not sure about the method name. first_depth_sensor seems to work for the depth sensor, but what if I want to set options like exposure and gain?