realsenseai/librealsense #1012: D400 device doesn't support depth streaming on macOS

gaborpapp, 2018-01-15:
| Required Info  |   |
|---|---|
| Camera Model | D400 | 
| Firmware Version |   05.08.15.00 | 
| Operating System & Version | macOS 10.12.6  |
| Kernel Version (Linux Only)  |   |
| SDK Version | SDK 2.0 git master 2a2446f1 and git development 3a5d701 |

error message from examples:
```
There are 1 connected RealSense devices.
unsupported descriptor subtype: 13
...
unsupported descriptor subtype: 13
The connected device doesn't support depth streaming!
```

The same camera works in Linux.

dorodnic, 2018-01-15:
What example are you running? Can you start streaming using the realsense-viewer?

gaborpapp, 2018-01-15:
Thanks for the quick reply. I tried the examples in `C` and `capture`. I also tried `realsense-viewer`.

It seems that I have to reconnect the camera several times. Sometimes it works. After one session I have to unplug and reconnect several times until I can make it work again. In Linux this is not required as far as I can tell.

dorodnic, 2018-01-15:
This is also what I experienced on the Mac. 
I'm still not sure if it is a setup issue (perhaps bad USB3 adapter) or an issue caused by the `libuvc` backend. 
One general limitation of `libuvc` backend is that whenever you kill the process under debug, you potentially leave the firmware in an undefined state (since there is no Kernel level driver to clean things up). In such situation, re-connect is unavoidable. Unfortunately, `libuvc` is currently the only way we know how to enable RealSense on Mac.

jnuevo, 2018-01-15:
I'm seeing the same issue too, using a D415 and SDK 2.9.0
In my system information tool, it appears as if the camera is connecting like a USB2 device. Is this expected?

> Intel(R) RealSense(TM) 415:
 Product ID:	0x0ad6
 Vendor ID:	0x8086  (Intel Corporation)
 Version:	50.8f
 Speed:	Up to 480 Mb/sec
 Manufacturer:	Intel(R) RealSense(TM) 415
 Location ID:	0x14200000 / 20

rsouthgate, 2018-01-19:
Also seeing the d415 connect as a usb2 device. There is no RGB camera available in the rs-viewer. Both infrared and depth streams are available though the depth is limited to 480x270 resolution.

dorodnic, 2018-01-19:
While we do have some stability issues on Mac, you should see the camera as USB3 device and have access to **all** functionality without exceptions. If the device is enumerated as USB2, you should inspect the cable and the USB dongle (if you are using one). Modern Mac-books come with a Renesas USB3 controller as far as I know, and it is top-notch.

rsouthgate, 2018-01-19:
No dongle - just a USB-c to USB-c cable (the one that came with the mbp)

tufeixp, 2018-01-25:
I have the same issue with my SR300 on MacBook Pro 2017 with a usb3 compatible dongle. 
It's able to stream RGB stream, but no depth stream!
I also tried R200, well, not work at all. Very frustrating!

[image]

gaborpapp, 2018-02-03:
I'm using the USB3 cable that came with the camera and it works fine most of the time. Sometimes it is indeed registered as USB2, but replugging the device solves it.

Although the depth stream resolution is stuck to the one streamed first on macOS.. If I start in 1280x720 and want to switch to 640x480 I have to replug the device to make this work. This works fine in Linux just by changing the resolution in `realsense-viewer`.

drhalftone, 2018-02-09:
I have the same issue with my D415.

mrjasonlu, 2018-04-04:
I can confirm I'm also having issues on my macbook pro 2012, sometimes the D415 connects as USB2 and sometimes USB3. 
Infrared and depth working but RGB stream crashes

arybalkin, 2018-04-04:
There seem to be plenty of issues related to USB3 implementation in DXXX cameras on  different hosts, not to mention what it's really difficult to find a good USB3 flexible cable working @ 5Gbit.  I hope Intel is listening to feedback from customers and will implement full USB2 support for all resolutions in the next firmware update.

gaborpapp, 2018-04-05:
@EladZucker tested the latest development branch and it seems to be much more stable.  Thank you!
`rs-capture` works well. `realsense-viewer` works with Custom presets, other presets throw an error about mulitple sensors (there's only one camera connected):
```
Invalid Value in rs2_open_multiple(sensor:0x7fb9f7e6e020, profiles:0x7fb9f7dc81f0, count:2):
get_xu(id=2) failed! Last Error: Operation timed out
```
C examples do not work for me. They get stuck after displaying device information.

Parisa-Rashidi, 2018-04-09:
I have the same issue with my D415 and MAC. 
unsupported descriptor subtype: 13
Is that a fix for this issue? 
I am using librealsense-2.10.2 and the example that I run is opencv_viewer_example.py

dorodnic, 2018-04-10:
Hi @Parisa-Rashidi 
The fix by @EladZucker should address the `unsupported descriptor` issue and some others, in addition we are looking into the Operation Timeout issue, planning to resolve it in our next release

JasonChang13, 2018-04-24:
hi @gaborpapp, @Parisa-Rashidi,

I've verified with 2.10.4 release with realsense viewer and play with all presets, all working fine, please let us know if this version fix your issue, thanks.

Regards,
Jason
Intel RSG Customer Platform Engineering

drhalftone, 2018-04-29:
This has gotten so much worse.  Now I can't get past rs2_pipeline_start_with_config(), it just completely freezes.  I've had to drop back to 2.10.2 just to get basic functionality working.  Is there any way I can pay one of your programmers to help me?

JasonChang13, 2018-04-30:
hi @drhalftone,

May I know your system configuration, error log and reproduce steps so we can help?
and maybe you can create a new ticket so we can focus on your case.

Regards,
Jason
Intel RSG Customer Platform Engineering

drhalftone, 2018-04-30:
See issue #1586, the function rs2_pipeline_start_with_config always hangs, even in the Intel Realsense supplied samples.  Here is my system:

  Model Name:	MacBook Pro
  Model Identifier:	MacBookPro11,2
  Processor Name:	Intel Core i7
  Processor Speed:	2.5 GHz
  Number of Processors:	1
  Total Number of Cores:	4
  L2 Cache (per Core):	256 KB
  L3 Cache:	6 MB
  Memory:	16 GB
  Boot ROM Version:	MBP112.0145.B00
  SMC Version (system):	2.18f15
  Serial Number (system):	C02NL0PZG3QN
  Hardware UUID:	DF02EA2F-187F-5E17-BAF8-083B693FCC08

and here is the output report after running cmake build:

DrHalftone:build dllau$ cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -G Xcode
-- The C compiler identification is AppleClang 9.1.0.9020039
-- The CXX compiler identification is AppleClang 9.1.0.9020039
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Info: REALSENSE_VERSION_STRING=2.10.4
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
CMake Warning at CMakeLists.txt:573 (message):
  Using libuvc!

-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2") 
-- Checking for one of the modules 'libusb-1.0'
-- Found OpenGL: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk/System/Library/Frameworks/OpenGL.framework   
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/dllau/SourceTree/LIBRealSense/build

I'm run a brew update and upgrade and am always running the latest Xcode release.

RealSense-Customer-Engineering, 2018-05-21:
[Realsense Customer Engineering Team Comment]
following this issue on #1586

RealSense-Customer-Engineering, 2018-05-24:
[Realsense Customer Engineering Team Comment]
hi @drhalftone,

per my test, adding sleep(1); after rs2_config_enable_stream, line 66 of rs-distance.c.
then it works, can you please check if this workaround also works for you?
thanks.

drhalftone, 2018-05-24:
It works on my Mac Book Pro, but not my Mac Pro.  On the Mac Pro, I need sleep(5).

RealSense-Customer-Engineering, 2018-05-25:
[Realsense Customer Engineering Team Comment]
hi @drhalftone,

thanks for confirm, please use this workaround while we are fixing this issue.
thanks.

dorodnic, 2018-07-12:
Proper fix is now in the development branch, will be released soon

dorodnic, 2018-07-22:
Everything should be resolved now in 2.14.0, if you encounter new problems with Mac OS, please open new issue.

closed 2018-07-22 by dorodnic

kmonte, 2018-07-24:
I'm using 2.14.0 and I still have an issue where when I kill the program I will still sometimes have to unplug/replug my d415 to get it functional again.