realsenseai/librealsense #1016: RealSense SDK 2.x installer does not include realsense2Config.cmake

UnaNancyOwen, 2018-01-15:
RealSense SDK 2.x installer (<code>Intel.RealSense.SDK.exe</code>) does not include <code>realsense2Config.cmake</code>.
Why does not includ config file in installer?

| Required Info  |   |
|---|---|
| Camera Model | SR300 / D400 | 
| Operating System & Version | Windows  |
| SDK Version | All Release of RealSense SDK 2.x |

Thanks,

dorodnic, 2018-01-15:
Yes, it would make sense to have it in the installer. 
The reason it was not bundled is simply because the installer was not really designed with CMake integration in mind. We offer Visual Studio `props` file to help configure C++ projects.

UnaNancyOwen, 2018-01-16:
I want to use CMake to generate Visual C++ project.
It is able to build user projects using same way in cross-platform. I think it is very modern way.
Please consider bundling the config file (<code>realsense2Config.cmake</code>) with installer.
Thanks,

ecarlsruh, 2018-02-26:
Where are the prop pages for the librealsense 2.10.0?  Also, do I have to build the libraries from the source code to compile and run the examples?

RealSense-Customer-Engineering, 2018-03-01:
[Realsense Customer Engineering Team Comment] 
@ecarlsruh
Please find v2.10.0 in https://github.com/IntelRealSense/librealsense/releases
If you use windows system, you can install **Intel.RealSense.SDK.exe to run the examples.**

RealSense-Customer-Engineering, 2018-03-01:
[Realsense Customer Engineering Team Comment] 
@UnaNancyOwen  Would like to know if you still need help on this question.

closed 2018-03-01 by UnaNancyOwen

dbolkensteyn, 2020-08-02:
Was this actually resolved @dorodnic ? I can't seem to find this file https://github.com/zivsha/librealsense/blob/getting_started_example/CMake/realsense2Config.cmake.in (without the `.in` suffix of course) after having installed the SDK on Windows.

Visual Studio 2019 has good out-of-the-box support for CMake now which makes it even more interesting then before as a cross-platform build tool.