realsenseai/librealsense #1019: Random execution exceptions

danielw256, 2018-01-16:
| Required Info  |   |
|---|---|
| Camera Model | SR300 & D400 | 
| Firmware Version | 3.21.0.0  | 
| Operating System & Version | Win10 Pro version 1709 build 16299.19  |
| SDK Version | 2.9.0 |

Using VS2017, C++. and COM DLLs, we have implemented the library. We are experiencing random execution exception crashes of 2 kinds (1) at the Windows message loop and (2) at a unknown code location (there is no debug info, even though all projects have PDB files and debug info).  This is a legacy robot program that used to use GigE cameras, so we know our code is very stable. These exceptions make no sense especially at the Windows message loop. How can the message loop be running fine, then suddenly, the message loop doesn't have execution access? Could something be changing the page tables? We have no code that does that. Only standard C++ and WinAPI.

dorodnic, 2018-01-16:
Hi @danielw256 
Are you using one of the pre-compiled releases? 
If yes, would you be willing to provide a crash-dump (or several dumps) to Intel for analysis? 
We are not aware of any memory corruption in the library. Also, the hourglass API pattern should be preventing any issues that might be related to ABI-compatibility. 
All our internal tests, including long-running ones are stable.
What is the reproduction rate you experience?

danielw256, 2018-01-16:
I'm using my own debug VS2017 build with default cmake settings. Tomorrow, I can switch and provide you with the dumps. The message loop dump probably won't be much help. Hopefully, the unknown line code dump will help. But I don't know why VS2017 wont show me the line of code where it crashed, since the realsense2.pdb is in the execution directory. Maybe it's crashing within Windows. I probably get a crash every 5-15 minutes, even after I finish using the camera using pipe.stop()

dorodnic, 2018-01-16:
Yes, this does sound strange indeed. We will be happy to take a second look at the dump file, hopefully we will be able to help. 
One experiment I can think of would be to comment out the content of `start` and `stop` [here](https://github.com/IntelRealSense/librealsense/blob/master/src/win/win-backend.cpp#L149).
The logic is - if `device_watcher` remains alive, it can respond to somewhat random events in the system, like connection of removable media. Of course, a. it shouldn't crash b. it should have provided a call stack and c. `device_watcher` is stopped and destroyed once pipeline / context go out of scope.

danielw256, 2018-01-17:
Where to upload the dump file?
It crashes both during camera use and after close.

dorodnic, 2018-01-18:
We don't have a good location. You can share it using google driver or Dropbox and send us the link (sergey.dorodnicov@intel.com)

danielw256, 2018-01-18:
@dorodnic Done. first 2 from windows message loop. 2nd 2 from unknown line.

dorodnic, 2018-01-19:
Got it, will update early next week.

dorodnic, 2018-01-22:
Hello @danielw256 
I have been looking into the files you provided, and so far can offer only very limited insight:
1. I think `evf4.dmp` was not captured with the same version of librealsense as the rest since it doesn't match symbols for 2.9.0
2. All crashes have threads related to realsense in the background, but as you pointed out the crash is not coming from them, and all they do is wait on some kernel objects. 
3. Some crashes seem to happen during streaming, some not during streaming. Please note that if you are properly disposing of all `rs2` objects, all `realsense` threads should disappear (synchronously). The fact that you still see `realsense` threads when not streaming means that some `rs2` objects are still around.  
4. Overall, I can confirm your observations regarding the message loop and the missing stack frame.
 
From here, I would suggest:

1. Could you please enable realsense logging into file and us a low level (DEBUG or INFO) log to help us understand how you work with the camera?
2. Could you please make sure that after streaming you destroy all realsense objects and check if the crash still occurs?

danielw256, 2018-01-25:
@dorodnic Is there a list of objects that are destroyed automatically by the C++ destructor and a list of objects that must be destroyed manually?

I reconfigured my thread to never stop snapping. So the camera was taking synchronized RGB and depth images for all 4 dumps.

danielw256, 2018-01-26:
@dorodnic I tried to make my thread synchronization better. It doesn't dump at the windows message loop anymore. Still dumps to an unknown location with no stack frame. I uploaded to your dropbox 4 more dump files with logging.

danielw256, 2018-01-29:
@dorodnic Crashes occur every 20 mins now.

dorodnic, 2018-01-30:
Hi @danielw256 
To answer your question on resource management - all RealSense resources are cleaned-up when the corresponding C++ object goes out of scope, there is no need for explicit clean-up. 
I saw the new files and looked over the logs, I will ask someone on the team to review the data for a second opinion.

aangerma, 2018-01-31:
Hi @danielw256 
I tried to load your crash dumps but they are not compatible with the symbols of 2.9.0,
can you please verify you are working with the official release?
If not can you reproduce with official release and send new dumps?

danielw256, 2018-02-01:
@aangerma I redid the dumps and logs with the 2.9.0 official release build. I uploaded them to @dorodnic dropbox

danielw256, 2018-02-01:
I

danielw256, 2018-02-28:
I think librealsense and COM are just incompatible.

closed 2018-02-28 by danielw256

danielw256, 2018-07-03:
We had some old / bad COM files in our repo. Sorry.