realsenseai/librealsense #1021: [D435] Recommended configuration for near-range (< 1m) accuracy

jfhamlin, 2018-01-16:
| Required Info  |   |
|---|---|
| Camera Model | D435 | 
| Firmware Version | 05.08.06.00 | 
| Operating System & Version | Ubuntu 14.04 |
| Kernel Version (Linux Only)  | 4.4.0-103-generic |
| SDK Version | 2.9.0 (current master@`2a2446f`) |

I'm testing out D435 for my use case, and sensor precision using default depth settings is much poorer than expected. The subject is a human face around 0.15m-1m away from the sensor, and the 3D result is invariably a blob with significant dynamic noise. I'd be happy to provide some example scans, but I'll begin the conversation here in case you all have some upfront recommendations.

dorodnic, 2018-01-17:
Hi @jfhamlin 
Could you please share your results (you can attach images by dragging them into comment text-box)?
Obviously, what one considers good is subjective but you should expect to see a dense point-cloud with clearly visible landmarks (nose, ears, eyes, mouth open or closed etc..)
Something like:
[image]
[image]
These are obviously not representative of the product as whole, but rather something I just captured with my D435 at around 30 cm range at moderate indoor lighting. 
The basic relevant settings are laser-power (should be on if you are indoor), auto-exposure on, visual preset and post-processing.
The first image was captured without post-processing with **Mid-Density** preset and the second with post-processing and **ShortRange** preset.

tchapi, 2018-01-29:
Hello @dorodnic 
I have a sensibly similar issue with a D435. I want to try your settings but as far as I understand, there is no "**Short Range**" visual preset for D400 series in [rs_option.h](https://github.com/IntelRealSense/librealsense/blob/ba01147d65db16fdf4da36a3e718fe81c8421034/include/librealsense2/h/rs_option.h), it only seems to be for SR300 series.

Can you tell us what specific option / visual preset you use for short range for the D435 ?

Is setting `RS2_RS400_VISUAL_PRESET_MEDIUM_DENSITY` sufficient ? I have tried to use `RS2_OPTION_MIN_DISTANCE` but it does not seems to be what I'm looking for (and it is not supported by the D435 apparently)

Thanks a lot !

icarpis, 2018-01-29:
Hello @tchapi, for a short range video capturing we have a [JSON](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets#short-range-preset) file that could help you on that.
In order to load the JSON file you can use the advanced mode API.
Here is a short example of how to do that:
```cpp
auto advanced_mode_dev = dev.as<rs400::advanced_mode>();
std::string json_content = read_the_json_file_content(...);
advanced_mode_dev.load_json(json_content);
```

tchapi, 2018-01-29:
Hi @icarpis !
All the JSON files have been removed from the repo unfortunately - do you have them somewhere else ?

Thanks

icarpis, 2018-01-29:
All the JSON files could be downloaded from [here](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets).
Each preset has a link(s) to its corresponding JSON file.

tchapi, 2018-01-29:
My bad — got them. Thanks again !

GitwellAnyohub, 2018-02-14:
Hey @jfhamlin , here are a list of parameters that I found effective at manipulating depth resolution at close range. I'm using the D415 camera and not the D435 so things might be slightly different.
In no particular order:
Exposure
Gain
Laser Power
DS neighbor threshold
Estimate mediam decrement
score maximum threshold 
depth units    
disparity shift
depth clamp max
depth clamp min

I'm currently working on this problem myself so if I find anything revolutionary I'll let you know.
Here is the thread I started about a similar issue
https://github.com/IntelRealSense/librealsense/issues/1171

RomRoc, 2018-03-07:
Hello, did anyone compare near range accuracy of D415 vs D435 with sample pictures?
I'm quite disappointed of my D435 accuracy, when I use 3D view I get strong noise and bad z-accuracy.
From pictures in [this page](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets) D415 seems to have better quality.
Thanks

RealSense-Customer-Engineering, 2018-04-18:
[Realsense Customer Engineering Team Comment] 
This document will be good for you to tune up your depth quality.
https://realsense.intel.com/wp-content/uploads/sites/63/BKMs-For-Tuning-RealSense_D4xx_Cameras_WP_1.7.pdf

jfhamlin, 2018-04-27:
@RealSense-Customer-Engineering thanks! I'm happy to have this issue closed for the time being, unless there are plans to expose better near range presets or more tuning suggestions for close-range use cases.

jfhamlin, 2018-04-27:
Though looking back at @dorodnic's results, I still see _considerably_ more depth noise at that range.

closed 2018-05-14 by RealSense-Customer-Engineering