realsenseai/librealsense #1027: Weird glitch SR300 depth aligned to color

Bardo91, 2018-01-17:
| Required Info  |   |
|---|---|
| Camera Model |  SR300| 
| Firmware Version |  3.21.0.0  | 
| Operating System & Version |   Ubuntu 16.04 |
| Kernel Version (Linux Only)  |  4.4.0-109-generic  |
| SDK Version |  realsense2 |

I am having a weird glitch while building a 3D point cloud from the aligned depth with the color stream. almost everything looks alright but there are some weird lines that coincide with some borders in the depth image. I recorded two GIFs to show it:

[image]

[image]

This is the code I am using:

```
auto list = mRsContext.query_devices();
if (list.size() == 0) {
	std::cout << "There's not any compatible device connected." << std::endl;
	return false;
}
auto mRsDevice = list[mDeviceId];

rs2::config cfg;
cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
auto mRsPipeProfile = mRsPipe.start(cfg);

auto depth_stream = mRsPipeProfile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
auto color_stream = mRsPipeProfile.get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>();

auto mRsAlign = new rs2::align(RS2_STREAM_COLOR);

auto mRsDepthToColor = depth_stream.get_extrinsics_to(color_stream);
auto mRsColorToDepth = color_stream.get_extrinsics_to(depth_stream);
auto mRsDepthIntrinsic = depth_stream.get_intrinsics();
auto mRsColorIntrinsic = color_stream.get_intrinsics();

auto sensor = mRsPipeProfile.get_device().first<rs2::depth_sensor>();
auto mRsDepthScale = sensor.get_depth_scale();

rs2::frameset frames = mRsPipe.wait_for_frames();

auto processedFrames = mRsAlign->proccess(frames);

rs2::frame frameDepth = processedFrames.first(RS2_STREAM_DEPTH);
rs2::frame frameRGB = processedFrames.first(RS2_STREAM_COLOR);

cv::Mat mLastRGB = cv::Mat(cv::Size(mRsColorIntrinsic.width, mRsColorIntrinsic.height), CV_8UC3, (void*)frameRGB.get_data(), cv::Mat::AUTO_STEP);
	

cv::Mat mLastDepthInColor = cv::Mat(cv::Size(mRsDepthIntrinsic.width, mRsDepthIntrinsic.height), CV_16U, (uchar*) frameDepth.get_data(), cv::Mat::AUTO_STEP);

pcl::PointCloud<pcl::PointXYZRGB> cloud
for (int dy = 0; dy < mLastDepthInColor.rows; dy = dy + mDownsampleStep) {
        for (int dx = 0; dx < mLastDepthInColor.cols; dx = dx + mDownsampleStep) {
				// Retrieve the 16-bit depth value and map it into a depth in meters
            uint16_t depth_value = mLastDepthInColor.at<uint16_t>(dy, dx);
				float depth_in_meters = depth_value * mRsDepthScale;
            // Set invalid pixels with a depth value of zero, which is used to indicate no data
            pcl::PointXYZRGB point;
            if (depth_value == 0) {
                if (mUseUncolorizedPoints) {
                    point.x = NAN;
                    point.y = NAN;
                    point.z = NAN;
                }
                else
                    continue;
            }
            else {
                // Map from pixel coordinates in the depth image to pixel coordinates in the color image
                cv::Point2f depth_pixel(dx, dy);
                cv::Point3f depth_point = deproject(depth_pixel, depth_in_meters);
                point.x = depth_point.x;
                point.y = depth_point.y;
                point.z = depth_point.z;
                auto rgb = mLastRGB.at<cv::Vec3b>(dy, dx);
                point.r = rgb[2];
                point.g = rgb[1];
                point.b = rgb[0];

            }

		cloud.push_back(point);
        }
    }
```

dorodnic, 2018-01-17:
Hi @Bardo91 
This is rather unusual. We will try to reproduce and help debug.
I assume you don't see this behavior using the RealSense Viewer?

Bardo91, 2018-01-17:
Hi @dorodnic,
Exactly, with the RealSense-viewer the point cloud looks perfect.
[image]
And I was working with RealSense_v1 before without any problem. It is a bit hard to see, but in the depth  this lines ca be seen. I rounded them in red:
[image]

Bardo91, 2018-01-17:
If I don't use the aligned frames, the glitch does not appear, i.e. if I use this:
```
rs2::frameset frames = mRsPipe.wait_for_frames();
rs2::frame frameDepth = frames.first(RS2_STREAM_DEPTH);
rs2::frame frameRGB = frames.first(RS2_STREAM_COLOR);

```
instead of this:

```
rs2::frameset frames = mRsPipe.wait_for_frames();
auto processedFrames = mRsAlign->proccess(frames);
rs2::frame frameDepth = processedFrames.first(RS2_STREAM_DEPTH);
rs2::frame frameRGB = processedFrames.first(RS2_STREAM_COLOR);
```

This are a pair of screenshots with the non aligned frames
[image]

[image]

But obviously, the colors are shifted as can be seen in the arm.

andyzeng, 2018-01-19:
I've been having a similar issue: strange line-shaped artifacts across the depth image aligned to color. There are still depth values on the lines, only they are inaccurate. This wasn't a problem a few weeks ago.

[image]

manhha1402, 2018-01-24:
I think this is the exact problem that I posted few weeks ago (https://github.com/IntelRealSense/librealsense/issues/925)

Bardo91, 2018-01-24:
@manhha1402  that's true, it is the same problem.

dorodnic, 2018-08-30:
This issue seem to have been fixed somewhere between 2.9.0 and 2.9.1, but the developer forgot to update on GitHub. Now everything looks good to me, but please open an issue if something like this appears again.

closed 2018-08-30 by dorodnic