realsenseai/librealsense #1028: optional installation location of python binding

marvin-hao, 2018-01-18:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version |  UBUNTU 16.04 |
| Kernel Version (Linux Only)  | 4.4.0-109-generic  |
| SDK Version | rs2 |

Is it possible to provide a customised value for LIBRARY DESTINATION when installing the Python wrapper? It's currently using CMAKE_INSTALL_LIBDIR but lib/pythonx.x/site-packages would be more preferable since sometimes we build python from source with different configurations.

Additionally, it'll be great if the author can mention two configurable variables PYTHON_INCLUDE_DIRS and PYTHON_LIBRARIES in README, which are provided by pybind11.

Thanks!

zivsha, 2018-02-03:
Hi @marvin-hao ,
Why do you need the pybind11 cmske variables? Is the SDK's `-DPYTHON_EXECUTABLE` not enough?

jlgarcia75, 2018-03-02:
Hello @marvin-hao,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

marvin-hao, 2018-03-02:
Hi @jlgarcia75,

Currently, we copy all binaries to site-packages manually. It makes more sense if it can install the Python wrapper in PYTHON_LIBRARIES like what OpenCV did, instead of CMAKE_INSTALL_LIBDIR.

Thank you!

RealSense-Customer-Engineering, 2018-04-13:
[Realsense Customer Engineering Team Comment] 
Hello @marvin-hao,

Thanks for your feedback, it has been forwarded to the engineering team. Closing issue.

Regards,
Jesus G.
Intel Customer Support

closed 2018-04-13 by RealSense-Customer-Engineering