realsenseai/librealsense #1030: I have ubuntu 14.04.5, but Kernel 4.4.0-109, can I install the latest version librealsense2 ?

summer1216, 2018-01-18:
| Required Info  |   |
|---|---|
| Camera Model |  SR300 | 
| Operating System & Version |  ubuntu 14.04.5  |
| Kernel Version (Linux Only)  | 4.4.0-109  |
| SDK Version | 2 |

Since I must use ROS-Indigo, I don't want to update to ubuntu 16.04. But for realsense I already updated my kernel to 4.4.0. I want to ask if this is ok for using the SDK2. Thanks!

dorodnic, 2018-01-18:
Hi @summer1216 
I believe everything will work. Please follow [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) to build the library from source, since on Ubuntu 14 the Debian package is missing `glfw` as far as I recall. 
Please update the issue if you encounter any problems

summer1216, 2018-01-18:
hi @dorodnic , thank you for your reply. Before I start to install, one more question please, I already installed **ros-indigo-librealsense** and **ros-indigo-realsense-camera** packages, which I think is the librealsense1 version. **Do I need to deinstall them** first to install the librealsense2, or it doesn't matter? Thanks!

icarpis, 2018-01-18:
You don't need to reinstall the `ros-indigo-librealsense` and `ros-indigo-realsense-camera` packages. The [ROS wrapper](http://wiki.ros.org/realsense_camera) for [SDK1](http://wiki.ros.org/librealsense) and the [ROS wrapper](https://github.com/intel-ros/realsense/tree/development) for [SDK2](https://github.com/IntelRealSense/librealsense/releases/) can live together.

summer1216, 2018-01-18:
@dorodnic @icarpis  It works!!!  I followed installation.md and tested with SR300 by the tools and examples, so far everything works fine. I can go on with developing now. Thank you very much for the kind tips.

dorodnic, 2018-01-19:
Closing this one, feel free to open more issues if you encounter any problems

closed 2018-01-19 by dorodnic