realsenseai/librealsense #1039: Install of librealsense-uvcvideo install on clean Ubuntu 16.04.03 Fails

jallard-sr, 2018-01-18:
| Required Info  |   |
|---|---|
| Camera Model | D415 | 
| Firmware Version |  Out of box | 
| Operating System & Version | Ubuntu 16.04.3 dist-upgraded on 1/18/18  |
| Kernel Version (Linux Only)  | 4.13.0-26-generic #29~16.04.2-Ubuntu SMP Tue Jan 9 22:00:44 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux  |
| SDK Version | 2.0 |

I attempted to install my new D415 onto a clean Ubuntu 16.04.03 desktop that had been dist-upgraded, and the installation failed. Here are the steps I followed, and I've attached the output from apt-get and the contents of the make.log file with the details.

I got my D415 out of the box, and followed the instructions from realsense.intel.com/develop/d400/get-started. I following the link to https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md, then configured the Realsense repository, registered the key, and ran ```sudo apt-get update```.

I then tried to install the first Realsense package using ``` sudo apt-get install realsense-uvcvideo```, and it failed. The output of the failure and the output of the failed compilation of uvc_driver.c are attached.

Any workarounds, fixes, or best suggestions to get this running in Linux?

Jim

[apt-realsense-uvcvideo-install.log](https://github.com/IntelRealSense/librealsense/files/1644333/apt-realsense-uvcvideo-install.log)
[uvc_driver_make.log](https://github.com/IntelRealSense/librealsense/files/1644334/uvc_driver_make.log)

dorodnic, 2018-01-18:
Hi @jallard-sr 
We will do our best to assist (this is most likely related to changes to Ubuntu 16 default kernel version).
In the meanwhile please note that you don't have to have `uvcvideo` package installed for basic depth streaming. You can proceed to the following steps of installing `librealsense2-utils` and should expect the tools to work.

jallard-sr, 2018-01-18:
Thanks for the quick response. I will try continuing with the librealsense2-utils package. Do you expect that the developer's packages, librealsense2-dev and librealsense2-dbg can also work without the realsense-uvcvideo package?

Cheers,

Jim

dorodnic, 2018-01-18:
Yes, the `uvcvideo` package contains a small number of patches to the Linux kernel, hence it is very sensitive to the kernel version. We still officially support only 4.4 and 4.10, my colleagues will provide information on how to work around this issue. 
These patches power mainly the time-stamp synchronization features, since hardware time-stamps are otherwise inaccessible from user-space.  
Rest of the library is kernel-agnostic

evanbeard, 2018-01-19:
Could you explain further what is and is not supported without the kernel mod? Do all of the APIs work fully without it? Or are they hampered in some way?

dorodnic, 2018-01-19:
To be precise, the `uvcvideo ` is responsible for the following features:
1. Access to hardware timestamp. `get_timestamp` will still work but you will get system-time instead. You can detect this at software level using `get_timestamp_domain` function. The patch to properly support this is at late review stages to be merged into next Linux kernel. 
2. Streaming with the SR300 camera, prior to kernel 4.13
3. Streaming with the D400 series, prior to kernel 4.4.0.59
4. Powerline-frequency camera control (part of UVC1.5 standard, not fully supported by Linux yet)

We provide Debian package with these patches for 4.4 and 4.10 kernels, but all the patches are also available under [scripts](https://github.com/IntelRealSense/librealsense/tree/master/scripts). Please see [Video4Linux backend preparation](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#video4linux-backend-preparation) section on how to apply kernel patches manually.

jallard-sr, 2018-01-19:
Before these comments, I did not know that you were only officially supporting 4.4 and 4.10 kernels, or that the uvcvideo patches were optional. I have spent time in the past attempting to run your scripts that apply patches to the uvcvideo, but it never succeeded for me. The scripts always failed in the final ```modprobe``` call to install the new module because of version number mismatches between the running kernel and the sources the script had downloaded.

If the issues with the module are going to remain until a future Ubuntu release, perhaps you could add some information in the installation instructions to let people know that the uvcvideo steps are optional. This could spare others the time that I spent attempting to make this work.

dorodnic, 2018-01-19:
I hope this information helps. 
Making kernel patches a thing of the past is our long term strategy, and we have already upstreamed several key patches. However, this process takes time.
In the meanwhile we are trying to:
a. Frequently rebase our patches on the latest, most relevant, kernel version. The switch from 4.10 to 4.13 as the default kernel for Ubuntu is very recent (was made earlier this year), so it caught us somewhat by surprise. 
b. Help community members troubleshoot manual patching (if you would like please open a separate issue on the patching problem and we will try to help)
c. Responding to community questions. This is not the first time we discuss publishing this information in an organised way, so this time I will make sure to escalate it.

jallard-sr, 2018-01-19:
I've been proceeding without the ```realsense-uvcvideo``` package, and so far so good. The utilities work, and I've been able to compile and run examples. I know it takes significant effort to make tested releases on so many platforms, and I for one really appreciate that on an LTS Ubuntu release I can connect to your repository and then apt just worked installing the rest of the packages. Thank you!

mpperez, 2018-02-08:
Same issue using Debian 4.9 and 4.14 kernels, apt output:
DKMS make.log for realsense-uvcvideo-4.10.0 for kernel 4.14.0-3-amd64 (x86_64)
vie feb  9 00:16:56 CET 2018
make: se entra en el directorio '/usr/src/linux-headers-4.14.0-3-amd64'
  AR      /var/lib/dkms/realsense-uvcvideo/4.10.0/build/drivers/media/usb/uvc/built-in.o
  CC [M]  /var/lib/dkms/realsense-uvcvideo/4.10.0/build/drivers/media/usb/uvc/uvc_driver.o
  CC [M]  /var/lib/dkms/realsense-uvcvideo/4.10.0/build/drivers/media/usb/uvc/uvc_queue.o
/var/lib/dkms/realsense-uvcvideo/4.10.0/build/drivers/media/usb/uvc/uvc_driver.c: In function ‘uvc_probe’:
/var/lib/dkms/realsense-uvcvideo/4.10.0/build/drivers/media/usb/uvc/uvc_driver.c:2065:12: error: ‘struct media_device’ has no member named ‘driver_version’; did you mean ‘driver_name’?
  dev->mdev.driver_version = LINUX_VERSION_CODE;
            ^~~~~~~~~~~~~~
            driver_name

driver_version member has been removed :
[https://github.com/torvalds/linux/commits/master/include/media/media-device.h](url)

RealSense-Customer-Engineering, 2018-03-06:
[Realsense Customer Engineering Team Comment] 
Hi @jallard-sr

Still need any further help for your build system?

Hi @mpperezH
With kernel 4.14 there should be no need for additional patches to stream D415. Our test configuration is Ubuntu 16.04.3 + kernel 4.14.15.

jallard-sr, 2018-03-06:
I do not require hardware timestamps on images, so I do not need the problematic uvcvideo patches. I'm successfully streaming color and depth on Ubuntu 16.04.4.

closed 2018-03-06 by jallard-sr

TheMikeyR, 2018-03-06:
I will just jump in and report that I had no issues installing `realsense-uvcvideo` on kernel `4.14.13-041413-generic` with Ubuntu 16.04.4
`realsense-uvcvideo is already the newest version (1.1.0-0ubuntu3).`

whatisor, 2019-07-25:
Hi, what is the difference between realsense-uvcvideo and librealsense2-dkms?
Thank you.