realsenseai/librealsense #1040: Examples are not working with connected camera. How to configure the default camera for example?

nawalgupta, 2018-01-19:
When I am running the any examples in windows, the app is crashing with unknown error. The examples are working when I point the camera cfg from load from file.

In the examples, how to connect to the default camera (cfg) is missing. I am not able to find any document using which I can connect to the default real sense camera.

The Real-Sense viewer is working fine with connected depth camera.

 My question is how to configure the default camera for examples or will it work without any changes?

```cpp
// Create a Pipeline - this serves as a top-level API for streaming and processing frames
rs2::pipeline p;

// Configure and start the pipeline
// My question is how to configure the default camera? or something is missing at this line?
p.start();

while (true)
{
}
```

| Required Info  |   |
|---|---|
| Camera Model |  D400 | 415 | 
| Firmware Version |   | 
| Windows 10 - Latest Update |   |
| Kernel Version (Linux Only)  |   |
| SDK Version | 2.0 (librealsense|

dorodnic, 2018-01-19:
Hello @nawalgupta 
The code snippet you provided should absolutely work. Please specify what error you are getting when running the app, and what operating system are you using.

nawalgupta, 2018-01-22:
I am using Windows 10 with latest update. It might be driver/hardware issue. I am able to access only Depth Stream in Real sense  Viewer.exe . 

Device manager is also showing only Depth Camera (no RGB Camera).

dorodnic, 2018-01-23:
Most likely the camera is identified as USB2 device and not USB3. This is not normal mode of operation for the device. In this situation, the camera will only offer certain depth resolutions but not RGB and most examples will not work.
Please make sure you are using the provided USB3 cable and you are connecting the device into a super-speed (blue) USB port on your platform.

nawalgupta, 2018-01-23:
Thanks. The issue is resolved after connecting to correct USB 3 port.

closed 2018-01-23 by nawalgupta