realsenseai/librealsense #1041: Packages missing from Linux repo

bhirashi, 2018-01-19:
On a clean install of Ubuntu 16.04.03, I followed the instructions here: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md.

My installation of `realsense-uvcvideo` failed as described in #1039, but also none of the librealsense2* packages are found.

`apt search realsense` returns only `realsense-uvcvideo`

dorodnic, 2018-01-19:
Hi @bhirashi 
Can you please double-check all the steps have completed successfully, including `sudo apt-get update`?
We have not yet uploaded the 2.9.1 Linux package, but as for the time of 2.9.0 release all packages has been present and tested. We will also check on our end.

bhirashi, 2018-01-19:
@dorodnic Yes, I'm sure I followed all the steps correctly. `realsense-uvcvideo` was downloaded and installed, and that would not have happened if I didn't do `sudo apt-get update`, right?

BjarneHerland, 2018-01-20:
FWIW, in my experience from setting up from sources I know the scripts patching the kernel to be slightly incomplete. In particular, the ubuntu-package "linux-headers-generic" is to be installed, but on my (fully updated) 16.04 system this package points to 4.4.0 linux-headers.

My workaround is to install the correct version of linux-headers and fetch the correct kernel-version from Ubuntu manually, not depending on the scripts for this. The rest of the patch-process works fine after this.

Perhaps a similar issue has slipped into the prebuilt binary packages? I was never able to make these work, which is why I started building everything from sources in the first place.

dorodnic, 2018-01-21:
The issue is most likely due to x86 version of Ubuntu. 
Our pre-built Linux packages are 64-bit only at this point (we will update the documentation)

RealSense-Customer-Engineering, 2018-02-28:
[Realsense Customer Engineering Team Comment] 
@bhirashi   Are you using x86 or x64 version of Ubuntu?

RealSense-Customer-Engineering, 2018-03-09:
[Realsense Customer Engineering Team Comment] 
@bhirashi  Would like to know if you still need help for this.

bhirashi, 2018-03-09:
I was using x86, so that was the problem. Thanks.

RealSense-Customer-Engineering, 2018-03-13:
[Realsense Customer Engineering Team Comment] 
@bhirashi  Could you please try x64 version?

closed 2018-03-19 by dorodnic