realsenseai/librealsense #1042: Are there any examples for gesture recognition on Linux?

mkhansen-intel, 2018-01-19:
Are there any examples either in librealsense or elsewhere for using Realsense for gesture recognition in Linux? I see this repo, but it is only for Windows: https://github.com/IntelRealSense/hand_tracking_samples

However that repo's readme says:
The code is primarily developed on Windows and includes Visual Studio solution files for VS2015 and VS2017 (hand_tracking_samples_vs2015.sln). **Linux and MacOS support is preliminary and compilation on these operating systems is not guaranteed.** (emphasis mine)

Can anyone point me to any sample code for using realsense for gesture recognition?

dorodnic, 2018-01-20:
Hi Matt,
Unfortunately, hand tracking sample is unfortunately just a POC at this point, we would love to make the effort to properly integrate it into our software stack later down the line.

mkhansen-intel, 2018-01-22:
@dorodnic - Thanks Sergey for the quick response, I'll continue to investigate options

closed 2018-01-22 by mkhansen-intel