realsenseai/librealsense #1043: pyrealsense problem with "fatal error: rs.h"

ljc19800331, 2018-01-20:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | The latest one  | 
| Operating System & Version | ubuntu 16.04  |
| Kernel Version (Linux Only)  |   |
| SDK Version | The latest one |

It seems that my librealsense was successfully built but I forgot to test an example and then install the pyrealsense. My installation process is shown in the following:

1.Linux installation
  a.sudo apt-get install git cmake3
  b.sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade
  c.I skip the process of "sudo update-grub && sudo reboot" and "uname -r"
  d.sudo apt-get install libudev-dev pkg-config libgtk-3-dev
  e.sudo apt-get install libglfw3-dev
  f.sudo apt-get install gcc-5 g++-5
  g.cmake ../ -DBUILD_EXAMPLES=true
  h.sudo make uninstall && make clean && make && sudo make install

2.pyrealsense installation
 a.git clone https://github.com/toinsson/pyrealsense.git
 b.cd pyrealsensse
 c.python setup.py install

Is my step correct? Do I need to add the environment path in linux for the .rs.h file? Thank you.

And get the following error when I want to "python setup.py install" for pyrealsense:

running install
running bdist_egg
running egg_info
writing requirements to pyrealsense.egg-info/requires.txt
writing pyrealsense.egg-info/PKG-INFO
writing top-level names to pyrealsense.egg-info/top_level.txt
writing dependency_links to pyrealsense.egg-info/dependency_links.txt
reading manifest file 'pyrealsense.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'pyrealsense.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
building 'pyrealsense.rsutilwrapper' extension
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/local/lib/python2.7/dist-packages/numpy/core/include -I/usr/local/include/librealsense -I/usr/include/python2.7 -c pyrealsense/rsutilwrapper.cpp -o build/temp.linux-x86_64-2.7/pyrealsense/rsutilwrapper.o
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
In file included from /usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarraytypes.h:1816:0,
from /usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/ndarrayobject.h:18,
from /usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/arrayobject.h:4,
from pyrealsense/rsutilwrapper.cpp:473:
/usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
#warning "Using deprecated NumPy API, disable it by "
^
pyrealsense/rsutilwrapper.cpp:476:16: fatal error: rs.h: No such file or directory
compilation terminated.
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

ljc19800331, 2018-01-20:
It shows that the librealsense is 100% finished, but I think the problem is from the installation of librealsense. But I don't know how to test it and find out the problem.

dorodnic, 2018-01-21:
Please note that [pyrealsense](https://github.com/toinsson/pyrealsense) only works with older versions of the project ([v1.12.1](https://github.com/IntelRealSense/librealsense/tree/legacy))
(`git checkout v1.12.1` before building librealsense)
In addition, we provide official [python wrapper](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python) for the latest versions of the SDK.

closed 2018-02-03 by zivsha

15151874182, 2019-03-07:
@dorodnic hi,i met the same problem when i try to run 'python setup.py install' on jetson TX2. I have not installed librealsense yet.
My question is:
Does pyrealsense package support the ARM architecture?
wish your help,thx.