realsenseai/librealsense #1044: Can't read from depth and color stream at same time in realsense viewer

IAmJulianAcosta, 2018-01-21:
| Required Info  |   |
|---|---|
| Camera Model | D400 | 
| Firmware Version | 5.08.15.0  | 
| Operating System & Version | Windows 10 16299 |
| SDK Version | 2.9.0 |

I described the error [here](https://communities.intel.com/thread/121760), and I'm copying and pasting:

> I just received a Intel D435, and tried the Realsense Viewer, but it seems that it can't read both streams from camera at the same time, the color stream is running at a maximum of 1fps and later the app says "no frames received" and the stream stops.
If I stop the depth stream, RGB stream starts again, but I can't make them work at the same time. It doesn't matter the resolution set in both streams, so I don't think that is a bandwidth issue.

The problem is that I opened the color stream using windows 10 camera app, and depth stream in realsense viewer, so I don't think this is a hardware issue.

Any ideas on this?

aangerma, 2018-01-21:
Hi,
Does the problem reproduce when synchronization between the streams is disabled? 
You can disable the synchronization by moving to 3D mode (in the top right corner), then click on the locker icon and finally back to 2D mode.
[image]

dorodnic, 2018-01-22:
I just want to confirm that even with the Sync disabled (the lock open) you are still getting "No Frames Received" for the RGB device in the viewer (but not in 3rd party application) 

During last two days, we worked under the assumption that the issue was due to the SDK having trouble to synchronize between Color and Depth because Color actual FPS was different from the requested one (requested: 30 most likely, actual: 15 in your screenshot) 

One way to understand what is going on is taking a look at the logs. Could you please run the viewer with librealsense logs at `DEBUG` level ([how to enable logs](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q&A#q-how-do-i-enable-librealsense-logs))?

jlgarcia75, 2018-02-27:
Hello @IAmJulianAcosta,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

IAmJulianAcosta, 2018-02-27:
Hey, sorry for the delay answering this. I couldn't find a way to reproduce it on a consistent basis, sometimes it just work, but it looks more like a hardware problem, when I reboot my laptop it works again. 

I'll try to reproduce it again so I can have a log

RealSense-Customer-Engineering, 2018-03-25:
[Realsense Customer Engineering Team Comment] 
Hello @IAmJulianAcosta,

Have you been able to repro this issue and acquire a log?

Regards,
Jesus
Intel Customer Support

closed 2018-04-13 by RealSense-Customer-Engineering