realsenseai/librealsense #1048: Cannot get depth frame by SR300. Error: Linux kernel not match.

KeCh96, 2018-01-22:
My OS is Ubuntu16.04, and I followed [the Linux tutorial](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md). When I run `realsense-viewer`, I get a GUI and I click the button for depth frame. But I get error:

> RuntimeError: The requested pixel format 'INZI' is not natively supported by the Linux kernel and likely requires a patch!
> Alternatively please upgrade to kernel 4.12 or later. Last Error: Inappropriate ioctl for device

My kernel is 4.10, so I upgrade to 4.13. Now I run `uname -sr` and get `Linux 4.13.0-26-generic`. But the error still occur. 

Note that I have installed the SDK on another laptop(also Ubuntu16.04) and no error occurred. So what caused this error? How to solve it? @zivsha

ev-mp, 2018-01-22:
Hi @KeCh96 , when installing librealsense on Ubuntu with kernel 4.10 you need to apply kernel patches either manually or with the dkms debian we provide.
We do not provide support for kernel 4.13 yet. Please switch to kernel 4.10 follow the instructions and publish kernel/application logs so we can look into your situation. Also, please specify whether you're using debian installation or build from source

ev-mp, 2018-01-22:
Hi @KeCh96 , after an additional thinking, kernel 4.13 should work out of the box for SR300 depth and infrared streaming.  Is it possible that you tried to apply the kernel patch on 4.13 and it failed? 
You can revert to stock version of the driver for 4.13, then capture the kernel log `dmesg -wT` while connecting SR300 so we'll get some details.

ev-mp, 2018-01-31:
Hello @KeCh96 , do you need more assistance on this?

dorodnic, 2018-02-26:
Closing for now, feel free to open new issue if you need further assistance.

closed 2018-02-26 by dorodnic