realsenseai/librealsense #1049: doesn't support architecture 'arm64'

newip, 2018-01-22:
| Required Info  |   |
|---|---|
| Camera Model | D400 | 
| Firmware Version |  05.08.15.00 | 
| Operating System & Version | Ubuntu 16.04 LTS (Nvidia TX2)  |
| Kernel Version (Linux Only)  | Linux tegra-ubuntu 4.4.38-tegra (Nvidia TX2) |
| SDK Version | not installed|

When step 2 doing apt-get update, the error show as below:
```
N: Skipping acquire of configured file 'main/binary-arm64/Packages' 
as repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease' 
#doesn't support architecture  #'arm64'
```

Then for install step, it shows unable to locate.
```
hostname:~$ sudo apt-get install realsense-uvcvideo
Reading package lists... Done
Building dependency tree      
Reading state information... Done
E: Unable to locate package realsense-uvcvideo
```

The device runs well in ubuntu PC. FYI. 

Please help to identify arm64 support issue. Thanks.

dorodnic, 2018-01-22:
Hello @newip 

We don't provide pre-built packages for ARM platform. You can however build the library from source (for example see [Installation on Jetson](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md))

newip, 2018-01-22:
Nice. Thank you @dorodnic 
Before seeing your update, I already successfully make it run according to [linux installation](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md).

Your note provide more information. It is helpful. Thanks.

closed 2018-01-22 by newip