realsenseai/librealsense #1050: SDK2.0 useless, lost many functions with pages

BBDXF, 2018-01-22:
| Required Info  |   |
|---|---|
| Camera Model | D400 | 
| Firmware Version | dont know  | 
| Operating System & Version |  win10  |
| Kernel Version (Linux Only)  |   |
| SDK Version | 2.0  |

i have forgot where the pages are,which use realsense devices to hand tracking, face recognition, skeleton tracking。i want to replace kinect with this one, but now it seems to be used only for depth data collection。is there any one who can tell me, what should i do?

BBDXF, 2018-01-22:
[RealSenseSDKdesignGuidelinesGold.pdf](https://github.com/IntelRealSense/librealsense/files/1651313/RealSenseSDKdesignGuidelinesGold.pdf)

just like this one.
i feel so sad~

gjmsoftware, 2018-01-23:
Hi @BBDXF , it is my understanding that the D415/D435 cameras do not have skeleton tracking like the Kinect. Furthermore, I'm not sure what functions, hand tracking and face recognition, from the previous SDK will be included. 

Currently, I don't believe any of those features are included, especially this early after launch.

jlgarcia75, 2018-01-24:
Hello @BBDXF,

RealSense focuses on providing depth across numerous platforms and wrappers.  We are fostering an open community to develop middleware. In addition, we are evaluating options with third parties and will announce more as we make the decision to adopt middleware for RealSense D400 series.

Regards,
Jesus
Intel Customer Support

BBDXF, 2018-01-24:
@jlgarcia75 @gjmsoftware i get it, thank you!

closed 2018-01-24 by BBDXF