realsenseai/librealsense #1051: project pixel from RGB to IR or depth issue

haoshuwen3, 2018-01-22:
| Required Info  |   |
|---|---|
| Camera Model | D415 | 
| Firmware Version |   | 
| Operating System & Version |   | Win10
| Kernel Version (Linux Only)  |   |
| SDK Version | |2.0

zivsha, 2018-01-22:
Hi @haoshuwen3 , can you please elaborate what exactly are you trying to achieve? 
You ask about color to depth alignment but your code uses infrared, color and depth frames..
We provide a `rs2::align` object that can do this for you, and you can take a look at its code as reference for your starting point

haoshuwen3, 2018-01-23:
Hi @zivsha , we had some points like eyes that were detected on RGB Image. Then we try to locate the corresponding points on IR image.  I took a look at the rs2::align code, and find the useful rsutil tool. But the projected points on IR is far from the real eye position.
Thanks

zivsha, 2018-01-23:
Can you please post a snippet that reproduces this?

zivsha, 2018-01-24:
Hi @haoshuwen3 ,

We've just pushed Pull Request #1068 which addresses the alignment of color to depth.
You can always deproject depth pixels (i,j), transform and project the result to the color image and copy that color pixel to the pixel (i,j) of the output image. See the changes in the PR to see how we do it.

travisWangXiaowei, 2018-02-09:
Hi @zivsha ,
what should i do If I want to get depth of some pixels in RGB frame? Since we cannot project pixel in color frame to depth frame.

zivsha, 2018-02-09:
The way to map a color pixel to depth pixel is by doing the same process as mapping depth to color, but returning the color pixels instead. The PR I mentioned uses this technique to calculate the entire frame,rather than a single pixel, as you have to through all depth pixels to find the one(s) matching the color pixel. If you use the `rs2::align` object, you can get the frames mapped to one another and then get the specific pixel.

travisWangXiaowei, 2018-02-10:
Hi  @zivsha,
I aligned Color frame to Depth frame, it seems that there is still offset between two frames.  Since depth frame and color frame are not spatially aligned, I should deproject all the pixels in depth frame , transform them and project them on color frame. Is that correct?

zivsha, 2018-02-10:
After using `rs2::align` with depth and color (either align depth to color or color to depth) the 2 imgaes you processed should spatially align, you don't have to de/project anymore, since this was performed on beach pixel during the align process. If you see any offset then that might be a bug and please share how you identify this offset. If you run `rs-align` example and make it align **to depth** do you see this issue?

RealSense-Customer-Engineering, 2018-03-02:
[Realsense Customer Engineering Team Comment] 
@haoshuwen3  **Would like to know if you still need help with this issue? **

closed 2018-03-27 by RealSense-Customer-Engineering

kargarisaac, 2018-07-02:
Hi,
How can I use align in python?
Thanks

mjfoo21, 2018-11-05:
Hello,

I know this has been closed but I can't use the `rs2::align ` as suggested.

My code is as followed:
```
rs2_stream align_to = RS2_STREAM_DEPTH;
rs2::align align(align_to);
rs2::frameset frames = pipe.wait_for_frames();
rs2::frameset processedFrames = align.process(frames);
rsFrame = frames.get_color_frame(); //this is the aligned depth frame
```

It can't get past the `align.process(frames);` line with an exception occurring at the `invoke()` under `class processing_block`:
> Exception thrown at 0x00007FFF61AA1F42 (realsense2.dll) in abc.exe: 0xC0000005: Access violation writing location 0x000001E9D6312128.

Actually I posted a question [a year ago](https://github.com/IntelRealSense/librealsense/issues/858) and it was fixed. Somehow it is not working now when I set up a new program and downloaded the repository few days ago.

I have looked at [this](https://github.com/IntelRealSense/librealsense/issues/2560) but it is for Matlab and also [this](https://github.com/IntelRealSense/librealsense/issues/1274) which suggesting calling the projecting functions myself.

I am using Windows 10 if it matters.