realsenseai/librealsense #1053: [D435] point cloud surface is not smooth

MorsLiang, 2018-01-22:
| Required Info  |   |
|---|---|
| Camera Model |  D435 | 
| Firmware Version | 05.08.06.00  | 
| Operating System & Version | Windows 10  |
| Kernel Version (Linux Only)  |   |
| SDK Version |RS2 |

I tested my newly bought Realsense D435, but I found that the quality of point cloud can't meet my requirements. Zoom in to local details,found It has a very obvious undulating wave and not a stable surface. Is that normal and how can i improve it? 

[image]

Zoom in to detail
[image]

dorodnic, 2018-01-22:
Depth generated from stereo can often suffer from this kind of noise. 
It really depends on:
1. Camera calibration (we have previously provided dynamic re-calibration application, and I believe such application is also available for the D400 series, but we will have to look-up the link)
2. Camera preset - if you are interested in depth accuracy you can prioritize it over fill rate using `High Accuracy` preset. Presets are available in the Viewer and also from code.
3. Post-processing - in recent versions of the SDK we offer several post-processing algorithms that reduce depth noise. You can enable and configure them individually in the Viewer or programmatically (we are about to publish an example that shows how to do this from code) 
4. The scene. You can still get depth from the D400 series even in very bad visual conditions (when laser has to be off, and the IR image is either over or under saturated), but you will get more noise. 

Based on the snapshots, I would recommend to first check presets and post-processing. 

We also provide Depth Quality app in this SDK that let's you play around with presets and see how different settings influence various metrics.

MorsLiang, 2018-01-23:
Hi @dorodnic thanks for your quick response. I had updated my SDK to latest and used post-processing algorithms trying to improve the quality of point cloud ,actually,it worked well.

but i had a problem when i applied the post-processing algorithms to my code such as The following:

auto decimate = std::make_shared<rs2::decimation_filter>();
decimate->set_option(RS2_OPTION_FILTER_MAGNITUDE, 2);
depth = decimate->process(depth);

auto depth_2_disparity = std::make_shared<rs2::disparity_transform>();
depth = depth_2_disparity->process(depth);
auto spatial = std::make_shared<rs2::spatial_filter>();
depth = spatial->process(depth);
auto temporal = std::make_shared<rs2::temporal_filter>();
depth = temporal->process(depth);
auto disparity_2_depth = std::make_shared<rs2::disparity_transform>(false);
depth = disparity_2_depth->process(depth);

the result point cloud was abnormal like this:
[image]
while the normal was 
[image]

and if i removed the decimation filter ,the point cloud's shape was normal but the quality was not the same as realsense-viewer's output, the following is realsense-viewer's output
[image]

What was wrong with my code or there was other problem that i didn't notice?

dorodnic, 2018-01-23:
Hi @MorsLiang 
I don't see anything obvious wrong, so I asked my colleague @ev-mp who implemented these filters to assist you. 
In addition, we are now (for some time) in the review stage of a dedicated [post-processing example](https://github.com/IntelRealSense/librealsense/pull/1010) (source code [here](https://github.com/AnnaRomanov/librealsense/blob/post_processing_demo/examples/post-processing/rs-post-processing.cpp)).

MorsLiang, 2018-01-24:
Hi @dorodnic 
Thanks a lot,your response is really helpful for me.
I has found the bug in my code, the decimate filter has reduced resolution of the depth image ,but i didnt notice this important information and i processed the depth image by the old camera intrinsic,so...

closed 2018-01-28 by ev-mp

stevensu1838, 2019-08-15:
Hi @dorodnic 
I am experiencing the [same problem](https://github.com/IntelRealSense/realsense-ros/issues/892)

agenis, 2021-02-16:
Hi @dorodnic thanks for this documentation on tuning the cameras. 
I also have problems of ondulations, with L515 and I would like to know if there are similar guidelines (or update) for L515, which is a different technology than the Dxxx products, so it might be different advices for tuning the camera?
thanks