realsenseai/librealsense #1056: Unable to retrieve intrinsics for infrared stream Y16/1280x800@25Hz

salovision, 2018-01-22:
| Required Info  |   |
|---|---|
| Camera Model | D435 | 
| Firmware Version |  05.08.15.00 | 
| Operating System & Version |  Windows 10 |
| Kernel Version (Linux Only)  |   |
| SDK Version | 2.0 |

Is there a reason why camera intrinsics can't be retrieved for infrared stream with 16-bit luminance?

Intrinsics can be retrieved for Y8 format but the image field of view seem to be a little different than with Y16 format.  Also, the image in Y16 format is sharper and less blurry than its Y8 counterpart. Any explanation for this?

dorodnic, 2018-01-22:
`Y16` for the D400 is defined as calibration (only) format and it has the property of not being rectified (not aligned to depth)
It is only supported when the device is in advanced mode and exposed only used for dynamic calibration. The SDK doesn't hide it if you want to use, but the hardware doesn't offer intrinsics for it.

salovision, 2018-01-22:
Okay, makes sense.

I believe this is still the only higher than 8-bit luminance format as the RGB sensor has stream modes with Y16 format but the content seem to be just scaled 8-bit data?

dorodnic, 2018-01-22:
Yes, you are correct (Y16 for the RGB camera is generated from same raw data as RGB8)

closed 2018-01-23 by salovision