realsenseai/librealsense #1058: ld: library not found for -lrealsense2

BoydRotgans, 2018-01-22:
Python build on Mac OSX issue, does anyone know what to do? Thanks a lot!

```
[machine-name]:build username$ make
Scanning dependencies of target pyrealsense2
[  5%] Building CXX object CMakeFiles/pyrealsense2.dir/python.cpp.o
[ 11%] Linking CXX shared library pyrealsense2.cpython-36m-darwin.so
ld: library not found for -lrealsense2
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [pyrealsense2.SOVERSION.cpython-36m-darwin.so] Error 1
make[1]: *** [CMakeFiles/pyrealsense2.dir/all] Error 2
make: *** [all] Error 2
[machine-name]:build username$ -v
-bash: -v: command not found
[machine-name]:build username$ make -v
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
```

zivsha, 2018-01-23:
Have you run the top level CMake (the one under the librealsense folder) before running `make`?

BoydRotgans, 2018-01-23:
@zivsha Yes this is what I did first. This was my output:
`cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true
-- The C compiler identification is AppleClang 9.0.0.9000039
-- The CXX compiler identification is AppleClang 9.0.0.9000039
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Found PythonInterp: /usr/local/bin/python3.6 (found version "3.6.4") 
-- Found PythonLibs: /usr/local/opt/python3/Frameworks/Python.framework/Versions/3.6/lib/libpython3.6m.dylib
-- Performing Test HAS_CPP14_FLAG
-- Performing Test HAS_CPP14_FLAG - Success
-- pybind11 v2.2.1
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- LTO enabled
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    BUILD_PYTHON_BINDINGS

-- Build files have been written to: /Users/[username]/build/librealsense-2.9.1_rc/wrappers/python/build
`

zivsha, 2018-01-24:
Hmm something starange is happening on your machien, I see you sepcified python biding flag, but cmake says it was ignored...
Can you try clean cmske cache and try again?

zivsha, 2018-02-03:
@BoydRotgans were you able to solve the issue?

djjacqmin, 2018-02-07:
After running `cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true`, I opened RealsensePythonWrappers.xcodeproj in the build/wrappers/python directory. In Xcode, I ran Build.

I believe that successfully built pyrealsense2.so and pybackend2.so in the the build/wrappers/python/Debug folder. I haven't successfully used them yet though (getting "No module named pyrealsense2" when I run the example scripts, but I think it is a problem with python not being able to find the files).

Have you tried the Xcode approach?

RealSense-Customer-Engineering, 2018-03-09:
[Realsense Customer Engineering Team Comment] 
hi @djjacqmin, @BoydRotgans,
would like to know if you still need help on this issue? thanks.

djjacqmin, 2018-03-09:
@RealSense-Customer-Engineering I have resolved my issue with re: "No module named pyrealsense2", simply had to add pyrealsense2 to path.

closed 2018-03-26 by dorodnic