realsenseai/librealsense #1059: QT Linking Issues

cardosoraph, 2018-01-22:
[project_file.txt](https://github.com/IntelRealSense/librealsense/files/1653355/project_file.txt)
| Required Info  |   |
|---|---|
| Camera Model |SR300 | 
| Operating System & Version | Windows 10  |
| SDK Version | 2.0|

Hi, I have compile the libraries from source in Windows 10 using VisualStudio. When I link the libraries in VisualStudio 2017, it works fine and runs the examples. However, I have created a simple qt project with an empty main function, where I'm only doing a "#include <librealsense2/rs.hpp>".

By doing that and hitting compile, I get many sintax errors from rs_frame.hpp, which don't really make any sense. I'm attaching my .pro file for reference.

Regards,
Raphael.

Jiloc, 2018-01-23:
Hello @cardosoraph, 
I think I already had this problem.
You have to include librealsense headers before any other QT header. This should be because QT uses a custom syntax for the foreach and it cause some problems with the one used in rs_frame.hpp

zivsha, 2018-01-23:
Hi @callendorph and @Jiloc ,
Can you please check if changing the `foreach(...)` in `rs_frame.hpp` into: `this->foreach(...)` resolves the issue without reordering the include lines?

Jiloc, 2018-01-23:
Unfortunately It doesn't, just to be clear, the line is [this](https://github.com/IntelRealSense/librealsense/blob/e68a256c308a66b98f94a0fe14e17d6e7cf5c29a/include/librealsense2/hpp/rs_frame.hpp#L655). Unfortunately also the `void foreach(T action) const` definition causes problems

zivsha, 2018-01-23:
Sounds strange as the functions is nested in the `rs2` namespace, is the QT foreach a macro? (Sorry for the hassle, I don't have a QT project to test this with)

UweDorsch, 2018-01-23:
The simplest way to solve this is to rename foreach in rs_frame.hpp.
There are only two places where you have to rename 'foreach' to something like 'foreachRS'

line 542
      template<class T>
        void foreachRS(T action) const
        {
      ...
        }

and line 484
       frame first_or_default(rs2_stream s) const
        {
            frame result;
            foreachRS([&result, s](frame f){
                if (!result && f.get_profile().stream_type() == s)
        ...

This will solve the Problem.

Jiloc, 2018-01-23:
@zivsha yes it is a macro

zivsha, 2018-01-23:
I see. Makes sense that it is a macro, same issue as the infamous min/max of windows.
We will consider changing the name of the function to avoid such issues in the future, thanks for the help guys!

zivsha, 2018-01-23:
@cardosoraph , I assume this will resolve your issue so I closed it. Please reopen if you are still having issues

closed 2018-01-23 by zivsha

daniel-packard, 2018-02-12:
I am working off master, and this is still a problem when trying to work with Qt5. Can we please keep this issue open until it is resolved by a PR -- or explicitly closed as a **wontfix**? 

+1  for resolving by changing the name of `foreach` in rs_frame.hpp

daniel-packard, 2018-02-13:
If this is part of a public interface that people outside the librealsense library are using, maybe we can disable the macro only within the region where it is being applied to librealsense: https://stackoverflow.com/questions/1543736/how-do-i-temporarily-disable-a-macro-expansion-in-c-c

bellekci, 2018-07-03:
foreach name clash with qt MACRO **still persists.**

Renamin librealse-defined foreach to foreachRS solves the problem.