realsenseai/librealsense #1060: Unable to locate package realsense-uvcvideo

mnissov, 2018-01-22:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 16.04 LTS  |
| Kernel Version (Linux Only)  | 4.4.38-tegra  |
| SDK Version | 2.0 |

Been using JetsonHacks' installer for Librealsense on the TX2 with performance issues so I've decided to follow the instructions at https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md.

The first error comes when doing `sudo apt-get update` wherein the error message:
```
W: Invalid 'Date' entry in Release file /var/lib/apt/lists/_var_libopencv4tegra-repo_Release
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease' doesn't support architecture 'arm64'
```

The main error coming with `sudo apt-get install realsense-uvcvideo`:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package realsense-uvcvideo
```

dorodnic, 2018-01-23:
Hi @mnissov 
At the moment we do not offer pre-compiled packages for the ARM platform, but you can build the library from source (please refer to [installation_jetson.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md))

closed 2018-02-04 by ev-mp

MohamedEhabHafez, 2019-03-31:
@dorodnic  is there the realsense2 pkg still not supported for 32 bit ubuntu??
and if yes is there another solution to install the pkg on the 32bit sys?