realsenseai/librealsense #1061: Only the depth module is recognized when the USB 3.0 cable is extended.

MatsumotoJ, 2018-01-23:
| Required Info  |   |
|---|---|
| Camera Model |  D435  | 
| Firmware Version | 0.508.15.00  | 
| Operating System & Version | Windows 10 (64 bit)  |
| Kernel Version (Linux Only)  |   |
| SDK Version | 2.9.0 |

Hello.

I am trying recently released D435 and D415 with Intel RealSense Viewer v2.9.0. The cameras are improved very much from the previous models. Thank you very much for your invention. 

The issue:
When I connected the camera with a USB 3.0 active extension cable (https://www.amazon.com/dp/B01FQ88CE6), the device manager recognize only depth module but not RGB module. Also, the Viewer call the connected device as Intel RealSense USB2 and can receive only low resolution depth streams (with a D415; a D435 does not work even for the low resolution depth). 

Previously the extension cable works well with a R200, as long as I used relatively low resolution Depth (320x240) and RGB (640x480) streams. I would like to use a D435 in similar setting, to exploit its advantage in the minimum distance.  

I need the extension for a multi-camera, 360 deg recording of aroud 1m3 space. 

If I could have an intermediate option (The device recognition as low resolution RGBD for relatively weak USB 3.0), it must be really helpful.

Thank you very much in advance.

WaterTian, 2018-01-23:
I am using the power line(USB-C) of MacBookPro to connected the camera .Also like this .

gjmsoftware, 2018-01-23:
I don't know if this will help, but when I first purchased the camera I had the exact same symptoms as you; the Viewer only recognized the depth stream and many of the settings were unavailable. 

My problem was that the camera was _not_ plugged into a USB 3 port. Once I plugged the camera into the correct USB3 port, the proper driver was installed and the viewer recognized the exact camera I had and the RGB stream was available. In my case, my motherboard had only 2xUSB3 ports and the other, more obvious USB ports wouldn't work. 

It is my prediction that your problem isn't with the cable extension, but the USB port you are using.

dorodnic, 2018-01-23:
I am still not 100% sure what would be a good answer - at the moment we don't have a recommended 3rd party extension cable. 
Seeing `... USB2` in the Viewer almost always indicate a setup problem. This is not something we have great deal of control over - the OS is the one who will ask for either USB3 or USB2 descriptors from the camera; once enumerated as USB2, the camera will not get sufficient bandwidth from the OS to offer both color and depth (it is possible that in future versions of firmware we will add some sort of low-res depth+color mode, but right now it is not supported)
Please try changing the port (as @gjmsoftware) suggested and consider trying a different cable.

MatsumotoJ, 2018-01-23:
Thank very much you for the info. 
The firmware support will be very helpful! Please consider to add the function!

I am sure that I used a USB 3.0 port.
The extension cable (https://www.amazon.com/dp/B01FQ88CE6) worked with R200 with low resolution (RGB: 640x480, Depth: 320x240). The extension cable was the best one among several different active repeater cables. Only this cable could work stably at that time. 

Meanwhile, I will try to find a good longer (around 3 m) USB3.0 cable for the direct connection without extension. I will let you know if I would find a good one.

WaterTian, 2018-01-24:
Thanks a lot,I will let you know if I would  find some other way .

MatsumotoJ, 2018-01-29:
Hi again,

The following 3m cable worked with D435 and D415: https://www.amazon.com/dp/B01MZIPYPY
I think that I can work with the cables for 360 degree recording by putting the PC below the floor of the recording space. 

However the firmware support is still very helpful for more flexible arrangement of devices.

Thanks!

alan-newnex, 2018-02-22:
Make sure your RealSense Camera is in USB3 state instead of USB2 when it's enumerated.
Checkout this video for the USB extension for RealSense D400 series camera: https://www.youtube.com/watch?v=GLQgR1jT04M

MatsumotoJ, 2018-02-22:
Hi, Thank you for the useful information. 

So you have special products designed for extending this type of USB3 cameras.

I have a few questions:
- Do your extension work as well as the short 1m cable? (i.e, no frame drop and no spontaneous disconnection of the link)
- If so, the stable connection could be kept for more than 2 hours of continuous recording?
- Do you have a Japanese distributor?

alan-newnex, 2018-02-22:
Hi Matsumoto,

Yes we have 1m A to C or 1m A to C Dual Screw Locking available.

You can directly buy those cables from our ecommerce website: https://www.ntcdistributing.com/a-to-c/

Or purchase through our Japanese Distributor: http://www.viewplus.co.jp

We have been manufacturing industrial grade USB 3.1 and 3.0 cable for many years. Our products supply industry like Machine Vision and Medical Imaging where continuous frame drop free image streaming is a must.

 

Alan Li
Associate Product Manager

Newnex Technology Corp.
3041 Olcott St
Santa Clara, CA 95054, USA
t: +1 (408)986-9988 ext 231
e: alanl@newnex.com

 <http://www.newnex.com/> www.newnex.com

 <https://www.facebook.com/Newnex-Technology-Corp-138785369498793/> Facebook icon  <https://twitter.com/Newnex> Twitter icon  <https://www.youtube.com/channel/UCOjDbQf-waOOX949CWTLPWQ> Youtbue icon

alan-newnex, 2018-02-22:
Hi Matsumoto,
Yes we do have 1m A to C or 1m A to C Dual Screw Locking available.
You can directly buy those cables from our ecommerce website: https://www.ntcdistributing.com/a-to-c/
Or purchase through our Japanese Distributor: http://www.viewplus.co.jp
We have been manufacturing industrial grade USB 3.1 and 3.0 cable for many years. Our products supply industry like Machine Vision and Medical Imaging where continuous frame drop free image streaming is a must.

alan-newnex, 2018-02-22:
Hi Matsumoto, we have stress tested all our solutions with the D435 camera. So far we haven't observed any frame drop issue. We have been manufacturing industrial grade USB3 connectivity solutions for many years. Our products supply industry like Machine Vision and Medical Imaging where continuous frame drop free image streaming is a must.
Right now since Type-C interface product is kind of new, you can directly purchase it from us through our website: www.newnex.com

MatsumotoJ, 2018-02-22:
Hi again,

Sorry I wanted to ask if your 20 m cable works as well as the 1m cable which is

MatsumotoJ, 2018-02-22:
Thank you for further information!
(Sorry for closing and opening the issue, it's just my errors)

jlgarcia75, 2018-03-02:
Hello @MatsumotoJ,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

MatsumotoJ, 2018-03-02:
Hello Jesus,

The firmware support for "some sort of low-res depth+color mode", suggested by dorodnic is still very helpful (The high quality cable above might be a bit expensive for people using our open source full-3D recording system with more than 4 cameras). 

I hope that you would consider adding the function to the firmware.

You can close the issue if you want.

Thank you,
Jumpei

TheMikeyR, 2018-03-09:
Anyone had any success using a 5m passive extension cable or active? newnex's site require an inquire and are not based in EU (so long shipping), I've considered to get this [cable](https://www.proshop.dk/USB-kabel/Pro-USB-30-Extension-AA-Black-5m/2466741) but if someone already tested a 5 meter passive and or active extender from a third-party, please share your experience!

RealSense-Customer-Engineering, 2018-04-17:
[Realsense Customer Engineering Team Comment] 
Thank you all for your suggestions for extended cabling, please consider this specific issue now closed.

If you wish to make a general posting for community input please consider using the community pages, https://communities.intel.com/community/tech/realsense. For @Q40, you may find more specific help on the Logitech support site.

If you have other specific issues or wish to follow up on this one, please feel free to open a new issue.

closed 2018-04-17 by RealSense-Customer-Engineering

srp-31, 2019-03-11:
> Hi again,
>
> The following 3m cable worked with D435 and D415: https://www.amazon.com/dp/B01MZIPYPY
> I think that I can work with the cables for 360 degree recording by putting the PC below the floor [...]

Hi
I am using a a D435 and would like to record color+depth stream 640x480 at 60fps. Atleast 30 fps. Could you please let me know the rates you got ?

MatsumotoJ, 2019-03-11:
Hi srp-31, 

Here the cable looks working for the setting with one D435:

[image]

Note I tested only for a few minutes.

srp-31, 2019-03-14:
Hi Matsumoto

Thank you for the quick reply. I had another camera (Logitech Brio) which has a long USB ccble. I was able to get the required performance with it.

npd26, 2019-05-10:
@TheMikeyR how was your experience with active extension cable? is it working?

TheMikeyR, 2019-05-15:
@npd26 Sorry for late answer, It didn't work with the extension cable I've linked above, but I've used a Deltaco USB3-1002 (rev 6) cable which is a 5m extension. Works great with no issues, and it gets power through the usb-port, so no need for external power.

npd26, 2019-05-15:
thank you for the answer. I am looking since 1 week in Germany, which cable will work with realsense D435.

fburak, 2020-08-04:
> 
> 
> @npd26 Sorry for late answer, It didn't work with the extension cable I've linked above, but I've used a Deltaco USB3-1002 (rev 6) cable which is a 5m extension. Works great with no issues, and it gets power through the usb-port, so no need for external power.

For others working with this issue, when I used D435 and D435i with the 5m "extension cable similar to you linked above", Realsense viewer sees cameras as connected from USB 2 port. However when i try to connect it with a powered usb hub (TP-Link 7-Port USB 3.0) it works well and viewer sees cameras connected as USB 3 port.