realsenseai/librealsense #1064: no response after launch ./scripts/patch-realsense-ubuntu-xenial.sh

duyilong, 2018-01-23:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 14.04  |
| Kernel Version (Linux Only)  | 4.4.0-111-generic  |
| SDK Version | |

I am following this [Linux Installation](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md) to install librealsense on my laptop, everything is OK until I ran 
`./scripts/patch-realsense-ubuntu-xenial.sh` 
After run this command, nothing happens.
I check the process properties using System Monitor, and I find out that the status of process "patch-realsense-ubuntu-xenial.sh"  is alway "Sleeping".
[image]

I have tried several times, same thing happens every time.

does anyone know why?

ev-mp, 2018-01-23:
Hi @duyilong , the script provides status message when completed - was it reporting success or failure in your case? 
Based on the kernel log only the  uvcmodule has been replaced, which is a positive indication.

Can you also post the results of  `dmesg -wT` when connecting the camera, and also the results of `lsusb` when the camera is already connected?

With regards to the "sleep"  - it indeed can be an issue with proxy settings, see more in [Troubleshooting chapter](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q&A#git)

duyilong, 2018-01-25:
@ev-mp Thank you for your enthusiastic reply. 
As I recall, the `patch-realsense-ubuntu-xenial.sh` script ran completely and terminal outputs show it has successfully install something which I didn't pay close attention.

I connect the camera, then pull it out, and then connect it again.
Then I get outputs for`dmesg | tail -50` as shown bellow:
>[duyl@ ~]14:37$ dmesg |tail -50
[ 1134.112757] usb 2-1: USB disconnect, device number 3
[ 1363.145712] usb 1-3.4: reset low-speed USB device number 6 using xhci_hcd
[ 1363.422340] usb 1-3.4: device firmware changed
[ 1363.422528] usb 1-3.4: USB disconnect, device number 6
[ 1363.681690] usb 1-3.3: reset low-speed USB device number 9 using xhci_hcd
[ 1363.960041] usb 1-3.3: ep 0x81 - rounding interval to 64 microframes, ep desc says 80 microframes
[ 1363.960046] usb 1-3.3: ep 0x82 - rounding interval to 64 microframes, ep desc says 80 microframes
[ 1364.029535] usb 1-3.4: new low-speed USB device number 10 using xhci_hcd
[ 1364.122105] usb 1-3.4: New USB device found, idVendor=1c4f, idProduct=0065
[ 1364.122108] usb 1-3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1364.122109] usb 1-3.4: Product: Usb Mouse
[ 1364.122111] usb 1-3.4: Manufacturer: SIGMACHIP
[ 1364.122191] usb 1-3.4: ep 0x81 - rounding interval to 64 microframes, ep desc says 80 microframes
[ 1364.124784] input: SIGMACHIP Usb Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.4/1-3.4:1.0/0003:1C4F:0065.0007/input/input27
[ 1364.177628] hid-generic 0003:1C4F:0065.0007: input,hidraw2: USB HID v1.10 Mouse [SIGMACHIP Usb Mouse] on usb-0000:00:14.0-3.4/input0
[ 2193.693984] systemd-hostnamed[4731]: Warning: nss-myhostname is not installed. Changing the local hostname might make it unresolveable. Please install nss-myhostname!
[ 2510.566086] systemd-hostnamed[4798]: Warning: nss-myhostname is not installed. Changing the local hostname might make it unresolveable. Please install nss-myhostname!
[ 2655.193988] usb 2-1: new SuperSpeed USB device number 4 using xhci_hcd
[ 2655.212657] usb 2-1: New USB device found, idVendor=8086, idProduct=0a80
[ 2655.212661] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2655.212663] usb 2-1: Product: Intel RealSense 3D Camera R200
[ 2655.212666] usb 2-1: Manufacturer: Intel Corp
[ 2655.212668] usb 2-1: SerialNumber: SN_2501000971
[ 2655.214710] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 2655.218178] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 2655.220992] uvcvideo: Unable to create debugfs 2-4 directory.
[ 2655.221103] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.2/input/input28
[ 2655.221756] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 2655.224618] uvcvideo: Unable to create debugfs 2-4 directory.
[ 2655.224723] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.4/input/input29
[ 2655.241311] systemd-udevd[4842]: failed to execute '/usr/local/bin/usb-R200-in_udev' '/usr/local/bin/usb-R200-in_udev': No such file or directory
[ 2655.255656] systemd-udevd[4849]: failed to execute '/usr/local/bin/usb-R200-in_udev' '/usr/local/bin/usb-R200-in_udev': No such file or directory
[ 2655.260677] systemd-udevd[4853]: failed to execute '/usr/local/bin/usb-R200-in_udev' '/usr/local/bin/usb-R200-in_udev': No such file or directory
[ 2715.598082] usb 2-1: USB disconnect, device number 4
[ 2741.334264] usb 2-1: new SuperSpeed USB device number 5 using xhci_hcd
[ 2741.352948] usb 2-1: New USB device found, idVendor=8086, idProduct=0a80
[ 2741.352951] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2741.352953] usb 2-1: Product: Intel RealSense 3D Camera R200
[ 2741.352954] usb 2-1: Manufacturer: Intel Corp
[ 2741.352956] usb 2-1: SerialNumber: SN_2501000971
[ 2741.354967] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 2741.358382] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 2741.361182] uvcvideo: Unable to create debugfs 2-5 directory.
[ 2741.361260] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.2/input/input30
[ 2741.361893] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 2741.364740] uvcvideo: Unable to create debugfs 2-5 directory.
[ 2741.364818] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.4/input/input31
[ 2741.380867] systemd-udevd[4891]: failed to execute '/usr/local/bin/usb-R200-in_udev' '/usr/local/bin/usb-R200-in_udev': No such file or directory
[ 2741.384133] systemd-udevd[4893]: failed to execute '/usr/local/bin/usb-R200-in_udev' '/usr/local/bin/usb-R200-in_udev': No such file or directory
[ 2741.384598] systemd-udevd[4895]: failed to execute '/usr/local/bin/usb-R200-in_udev' '/usr/local/bin/usb-R200-in_udev': No such file or directory
[duyl@ ~]14:37$ 

Besides,
here is the `lsusb` result after I plug the camera into the USB port. The first line shows my R200 is indeed connected, because before plugging R200, there is no current first line.
> [duyl@ ~]14:03$ lsusb 
Bus 002 Device 003: ID 8086:0a80 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 138a:0011 Validity Sensors, Inc. VFS5011 Fingerprint Reader
Bus 001 Device 007: ID 8087:0a2a Intel Corp. 
Bus 001 Device 005: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 003: ID 1bcf:2c01 Sunplus Innovation Technology Inc. 
Bus 001 Device 006: ID 1c4f:0065 SiGma Micro 
Bus 001 Device 004: ID 1c4f:0002 SiGma Micro Keyboard TRACER Gamma Ivory
Bus 001 Device 002: ID 05e3:0606 Genesys Logic, Inc. USB 2.0 Hub / D-Link DUB-H4 USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**How can I troubleshoot the problem based on those information?**

ev-mp, 2018-01-25:
Hello @duyilong , unfortunately I haven't noticed it earlier, but you're using R200 device which is supported by Librealsense version 1 on the `legacy` branch as [we advertise](https://github.com/IntelRealSense/librealsense/tree/development#overview).

Note that while that release in a "freeze" state we maintain the branch and provide support.

duyilong, 2018-01-26:
@ev-mp 
Thanks for point out the problem for me. 
Before getting start to install version 1, I would like to ask one more question.
Will  these operations already implemented affect the installation of version 1 ?

ev-mp, 2018-01-28:
Hello, the executable file you are trying to run is part of SDK v2,  the hint is the `rs2_...` prefix in the error messages:
> [duyl@ /usr/local/bin]14:43$ ./rs-capture
RealSense error calling rs2_pipeline_start(pipe:0xa00d50):

The executable files for librealsense v1 also have a different naming convention:
- **cpp**-capture/**cpp**-enumerate-devices/**cpp**-config-ui
- **c**-tutorial-.../**cpp**-tutorial-.../...

Since you're using R200 exclusively, I'd suggest to uninstall SDK2 to mitigate the chance of mixing compilation/execution of different SDK versions

ev-mp, 2018-01-31:
Hello @duyilong , I'm closing it as the root cause was identified and clarified.
Feel free to reopen this ticket/new one if you need more assistance with the setup.

closed 2018-01-31 by ev-mp