realsenseai/librealsense #1066: ImportError: No module named pyrealsense2

mikeNike9, 2018-01-23:
| Required Info  |   |
|---|---|
| Camera Model | D415 | 
| Firmware Version |   | 
| Operating System & Version |  Linux ubuntu 16.04LTS|
| Kernel Version (Linux Only)  |  Linux mikeNike9 4.4.0-109-generic |
| SDK Version | Intel® RealSense™ SDK 2.0  |

I have followed countless examples including all the directions that have been given inside the issues.
issue #793 
Followed these commands last week and it successfully worked on another dev machine.
I am now trying to do the same with a new machine and I get the follow error.
ImportError: No module named pyrealsense2

here is my directory  
/librealsense/build/wrappers/python 
CMakeFiles
cmake_install.cmake
depth.py
Makefile
pybackend2.cpython-35m-x86_64-linux-gnu.so
pybackend2.cpython-35m-x86_64-linux-gnu.so.2
pybackend2.cpython-35m-x86_64-linux-gnu.so.2.9.1
pyrealsense2.cpython-35m-x86_64-linux-gnu.so
pyrealsense2.cpython-35m-x86_64-linux-gnu.so.2
pyrealsense2.cpython-35m-x86_64-linux-gnu.so.2.9.1
third_party

can you please provide some assistance 

Thanks bros

mikeNike9, 2018-01-24:
No I just figured it out actually.
I had the wrong directory for anaconda3.
Thank you we can close this issue.
I ended up add this command on the cmake ..

 -DPYTHON_EXECUTABLE=/home/mike/anaconda3/bin/python3

closed 2018-01-24 by mikeNike9

kat-mcg-lin, 2019-05-27:
I have also been trying to get pyrealsense2 installed and use it in pycharm, however whenever I try to import pyrealsense in pycharm I get the following error:

`ModuleNotFoundError: No module named 'pyrealsense2' `

I have the same items in my python wrapper directory- how exactly did you get it to work? 

I have read many different issue threads and nothing seems to help

alisharifara, 2020-02-07:
is there any update about this issue? 
I am also getting the same error: 
ModuleNotFoundError: No module named 'pyrealsense2'

darmin-hsiung, 2020-11-08:
hello, i have a same question as follow, please help me to solove it. thanks.
$ python prs2.py 
Traceback (most recent call last):
  File "prs2.py", line 1, in <module>
    import pyrealsense2 as rs
  File "/home/tt/.local/lib/python3.6/site-packages/pyrealsense2-2.39.0.2342-py3.6-linux-x86_64.egg/pyrealsense2/__init__.py", line 2, in <module>
ModuleNotFoundError: No module named 'pyrealsense2.pyrealsense2'