realsenseai/librealsense #1067: When will the unity version of the SDK be released?Do you have a specific timetable?

zhengtiesheng, 2018-01-24:
| Required Info  |   |
|---|---|
| Camera Model |  D400 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |
| SDK Version | Unity3d|

When will the unity version of the SDK be released?Do you have a specific timetable?

dorodnic, 2018-01-24:
Hello @zhengtiesheng 

I believe this feature was communicated to customers as post-launch. 
In practice the team is working to enable it as soon as possible. We are trying to be transparent regarding any delay or problems, update on progress and be otherwise remain responsive to community feedback. Like with other wrappers, it is reasonable to assume that within one sprint we can start posting draft versions for review, and gradually add functionality over time. 

Please take into account that D400 product line does not include the same set of high-level middlewares as our previous products. Currently we are looking into mapping D400 streams to Unity textures using the .NET wrapper, next in line is point-cloud generation for visualization and for collision detection. 

Could you share what functionality you would like to see first within the Unity plug-in?

tufeixp, 2018-01-25:
Really wish for the feature of collision detection (spatial mapping) in unity, both on Mac and PC.

zhengtiesheng, 2018-01-25:
Thank you for your reply.        @  dorodnic
It is very good to use the point-cloud generation for visualization and for collision detection.Look forward to its early release.

marklundin, 2018-01-25:
Work from the ground up. As a start it would be great to have the feeds/streams available as textures

zivsha, 2018-01-31:
@zhengtiesheng , @tufeixp @marklundin your feefback on PR #1102 would be appreciated

tufeixp, 2018-02-05:
Fantastic!

closed 2018-02-26 by dorodnic