realsenseai/librealsense #1069: What's the recommended way to clear advanced mode on a D415?

jallard-sr, 2018-01-24:
| Required Info  |   |
|---|---|
| Camera Model | D415 | 
| Firmware Version | 05.08.15.00  | 
| Operating System & Version | Ubuntu 16.04.03  |
| Kernel Version (Linux Only)  | #34~16.04.1-Ubuntu SMP Fri Jan 19 17:11:01 UTC 2018  |
| SDK Version | 2.9.1-0~realsense0.59 |

I've read that advanced mode is sticky in the D415, and will persist on the device once enabled. I was experimenting in realsense-viewer with depth detection thresholds, and now rs-enumerate reports "Advanced Mode: YES", even after power cycling the device, as advertised.

I want to go back to experimenting with streaming in the default mode. The rs2::device::hardware_reset() method looked like it might do the trick.

What's the recommended way to clear advanced mode? And is there any harm in clearing it every time I start up?

Cheers,

Jim

zivsha, 2018-01-24:
Hi @jallard-sr ,
You can turn on and off advanced mode using the RealSense Viewer.
Depending on the version you run, it's either under the "More" menu icon, or under "Advanced Mode" menu of the Stereo Sensor

dorodnic, 2018-01-24:
Hi @jallard-sr 
We provide such option in the Viewer. Prior to 2.9.1 it was under `Controls > Advanced Mode > Disable Advanced Mode`. In the upcoming 2.9.1 it will be a checkbox under `More` button you can un-check.
If you mean how to do this from code, it would be something like:
```cpp
auto dev = context().query_devices().front();
auto adv = dev.as<rs400::advanced_mode>();
adv.toggle_advanced_mode(false);
adv.hardware_reset(); // I am not sure if the previous command will do this
```

icarpis, 2018-01-25:
@jallard-sr, regarding the code-snippet provided by @dorodnic, there is no need to do hardware reset (adv.hardware_reset()) after setting the advanced-mode to false because it's already done by toggle_advanced_mode(...) function.
Also, it is not harm to reset the advanced mode controls, but be aware that hardware reset will reset also the camera configuration such as controls, streaming and etc.

jallard-sr, 2018-01-25:
@icarpis When my application starts up, it is my intent to reset the camera controls, streaming, etc, to all of their defaults. Of course I want to retain the calibration and tuning for the device, but other than that I want to always start the camera like it just came out of the box. This way I can ensure that the camera is set up the way this particular application requires, and not be in a state that was useful to the last application.

The code snippet from @dorodnic was  what I was looking for, though I have added a call to ```adv.is_enabled()``` before entering that block to not take extra startup time when it's not needed. I'll remove the call to ```hardware_reset`` when clearing advanced mode.

What your comment brings up is the question of whether there is other state, beyond advanced mode, that I may need to reset to ensure the camera is in a default state. I'm guessing that ```hardware_reset``` clears all volatile state, but I don't know the extent of non-volatile state/setup that I may need to revert.

The broader questions I should have asked first are the following: Is there is a method that can tell me if the camera is already in its default settings, and are there methods beyond ```toggle_advanced_mode``` and ```hardware_reset``` that are needed to get the camera back into its default state?

icarpis, 2018-01-26:
There is no camera state that `hardware_reset` doesn't clear up. `hardware_reset` doing the same effect that physically unplug and plug back the camera does. So, `hardware_reset` is what you are looking for.
Regarding the additional questions, there is no method that can tell you if the camera in its default settings, and also, there are no other methods beyond `toggle_advanced_mode` and `hardware_reset` that get the camera back into its default state.

dorodnic, 2018-01-28:
One notable exception might be RGB controls - if I recall correctly, they persist between power-cycles.

icarpis, 2018-02-22:
I'm closing this issue. please raise any questions or new issues if you will have any further questions.

closed 2018-02-22 by icarpis