realsenseai/librealsense #1072: UVCVideo modprobe error: SR 300, build from source, kernel 4.4.0-98

Haveon, 2018-01-24:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | ?  | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  | 4.4.0-98-generic  |
| SDK Version | 2.0 (master branch, Jan 24) |

I'm having an issue building the SDK from source with the Python bindings. I had managed to successfully do it under kernel version 4.13.0-31-generic but due to other reasons (CUDA 9.1 installation) I ended up downgrading to 4.4.0-98-generic.

I'm following the instructions in https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md with a couple of modifications:

  - Did not perform ```sudo apt-get dist-upgrade``` or ```sudo update-grub && sudo reboot``` since the newest 4.4.0-xxx kernel seems to have issues with the newest nvidia driver (could be wrong about that though)
  - inside the build directory ran ```cmake ../ -DBUILD_EXAMPLES=true -DBUILD_PYTHON_BINDINGS=bool:true```
  - ran ```make -j8```
  - Changed line 23 of patch-realsense-ubuntu-xenial.sh to ```sudo apt-get install linux-headers-$(uname -r) build-essential git``` since I'd want to install the headers for my kernel

The error shows up when running the patch-realsense-ubuntu-xenial.sh script. It runs almost to completion, but near the end gives the following error:
 ```
Handle uvcvideo :
Applying the patched module ... modprobe: ERROR: could not insert 'uvcvideo': Invalid argument

Failed to insert the patched module. Operation is aborted, the original module is restored

Verify that the current kernel version is aligned to the patched module version

modprobe: ERROR: could not insert 'uvcvideo': Invalid argument
```

If I comment out line 119 (try_module_insert uvcvideo ...), the script manages to finish, but clearly uvcvideo either wasn't restarted or wasn't patched. Trying to run realsense-viewer results in the application not finding any cameras connected.

I used synaptic to download the linux-image for 4.4.0-98, and it's headers. I did manage to forget to include linux-image-extras at one point, which resulted in other errors, so I would not rule out other kernel packages I could have missed.

Any ideas?

ev-mp, 2018-01-25:
Hi @Haveon, the kernel patch is configured to use the latest version for the Ubuntu LTS sources, 4.4.0-114 or later in your case, that's why the instructions are to `sudo apt dist-upgrade` before applying the patch. 
Compiling the driver with 4.4.0-114 or later but applying it on 4.4.0-98 active kernel would probably fail due to symbols mismatch  - as highlighted in the error message you receive.

In case you have you use not the latest LTS kernel then the options are:
1. Install the dkms package we provide that will patch 4.4.0-98 only - this should be the fastest way.
   Follow the [Linux Installation Guide](https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md#installing-the-packages) and apply `sudo apt-get install realsense-uvcvideo` to get dkms.
2. Manually adjust the kernel patch to fetch a specific tag within Ubuntu github tree (4.4.0-98) instead of the recommended/latest.

Haveon, 2018-01-25:
I followed the instructions in the link provided. When I ran ```sudo apt-get install realsense-uvcvideo``` I got the following:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  realsense-sdk-udev-rules
The following NEW packages will be installed:
  realsense-sdk-udev-rules realsense-uvcvideo
0 upgraded, 2 newly installed, 0 to remove and 9 not upgraded.
Need to get 0 B/84.9 kB of archives.
After this operation, 838 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Selecting previously unselected package realsense-sdk-udev-rules:amd64.
(Reading database ... 261140 files and directories currently installed.)
Preparing to unpack .../realsense-sdk-udev-rules_2.7.7-0~realsense0.29_amd64.deb ...
Unpacking realsense-sdk-udev-rules:amd64 (2.7.7-0~realsense0.29) ...
Selecting previously unselected package realsense-uvcvideo.
Preparing to unpack .../realsense-uvcvideo_1.1.0-0ubuntu3_all.deb ...
Unpacking realsense-uvcvideo (1.1.0-0ubuntu3) ...
Setting up realsense-sdk-udev-rules:amd64 (2.7.7-0~realsense0.29) ...
Setting up realsense-uvcvideo (1.1.0-0ubuntu3) ...
Loading new realsense-uvcvideo-4.4.0 DKMS files...
First Installation: checking all kernels...
Building for 4.4.0-98-generic and 4.13.0-31-generic
Building initial module for 4.4.0-98-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/realsense-uvcvideo.0.crash'
Error! Bad return status for module build on kernel: 4.4.0-98-generic (x86_64)
Consult /var/lib/dkms/realsense-uvcvideo/4.4.0/build/make.log for more information.
Loading new realsense-uvcvideo-4.10.0 DKMS files...
First Installation: checking all kernels...
Building for 4.4.0-98-generic and 4.13.0-31-generic
Building initial module for 4.4.0-98-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/realsense-uvcvideo.0.crash'
Error! Application of patch 10-realsense-camera-formats-4.4.patch failed.
Check /var/lib/dkms/realsense-uvcvideo/4.10.0/build/ for more information.
```

ev-mp, 2018-01-26:
Hi @Haveon , can you provide the log file `'/var/crash/realsense-uvcvideo.0.crash'` ? Thanks

Haveon, 2018-01-26:
```
ProblemType: Package
DKMSKernelVersion: 4.10.0-42-generic
Date: Thu Jan 18 15:01:50 2018
Package: realsense-uvcvideo 1.1.0-0ubuntu3
PackageVersion: 1.1.0-0ubuntu3
SourcePackage: realsense
Title: realsense-uvcvideo 1.1.0-0ubuntu3: realsense-uvcvideo kernel module failed to build
ApportVersion: 2.20.1-0ubuntu2.15
Architecture: amd64
Dependencies:
 binutils 2.26.1-1ubuntu1~16.04.5
 coreutils 8.25-2ubuntu3~16.04
 cpp 4:5.3.1-1ubuntu1
 cpp-5 5.4.0-6ubuntu1~16.04.5
 dkms 2.2.0.3-2ubuntu11.5
 fakeroot 1.20.2-1ubuntu1
 gcc 4:5.3.1-1ubuntu1
 gcc-5 5.4.0-6ubuntu1~16.04.5
 gcc-5-base 5.4.0-6ubuntu1~16.04.5
 gcc-6-base 6.0.1-0ubuntu1
 kmod 22-1ubuntu5
 libacl1 2.2.52-3
 libasan2 5.4.0-6ubuntu1~16.04.5
 libatomic1 5.4.0-6ubuntu1~16.04.5
 libattr1 1:2.4.47-2
 libc-dev-bin 2.23-0ubuntu10
 libc6 2.23-0ubuntu10
 libc6-dev 2.23-0ubuntu10
 libcc1-0 5.4.0-6ubuntu1~16.04.5
 libcilkrts5 5.4.0-6ubuntu1~16.04.5
 libfakeroot 1.20.2-1ubuntu1
 libgcc-5-dev 5.4.0-6ubuntu1~16.04.5
 libgcc1 1:6.0.1-0ubuntu1
 libgmp10 2:6.1.0+dfsg-2
 libgomp1 5.4.0-6ubuntu1~16.04.5
 libisl15 0.16.1-1
 libitm1 5.4.0-6ubuntu1~16.04.5
 libkmod2 22-1ubuntu5
 liblsan0 5.4.0-6ubuntu1~16.04.5
 libmpc3 1.0.3-1
 libmpfr4 3.1.4-1
 libmpx0 5.4.0-6ubuntu1~16.04.5
 libpcre3 2:8.38-3.1
 libquadmath0 5.4.0-6ubuntu1~16.04.5
 libselinux1 2.4-3build2
 libstdc++6 5.4.0-6ubuntu1~16.04.5
 libtsan0 5.4.0-6ubuntu1~16.04.5
 libubsan0 5.4.0-6ubuntu1~16.04.5
 linux-libc-dev 4.4.0-109.132
 lsb-base 9.20160110ubuntu0.2
 make 4.1-6
 manpages 4.04-2
 manpages-dev 4.04-2
 multiarch-support 2.23-0ubuntu10
 patch 2.7.5-1
 realsense-sdk-udev-rules 2.7.7-0~realsense0.29 [origin: Freight]
 zlib1g 1:1.2.8.dfsg-2ubuntu4.1
DistroRelease: Ubuntu 16.04
InstallationDate: Installed on 2017-11-19 (60 days ago)
InstallationMedia: Ubuntu 16.04.3 LTS "Xenial Xerus" - Release amd64 (20170801)
JournalErrors:
 -- Logs begin at Fri 2018-01-05 12:50:27 EST, end at Thu 2018-01-18 15:02:04 EST. --
 Jan 05 12:50:27 Neuron8 kernel: ENERGY_PERF_BIAS: Set to 'normal', was 'performance'
 Jan 05 12:50:27 Neuron8 kernel: ENERGY_PERF_BIAS: View and update with x86_energy_perf_policy(8)
 Jan 05 12:50:27 Neuron8 kernel: PCCT header not found.
 Jan 05 12:50:27 Neuron8 kernel: rtc rtc0: invalid alarm value: 2018-1-5 17:60:25
 Jan 05 12:50:27 Neuron8 kernel: nvidia: loading out-of-tree module taints kernel.
 Jan 05 12:50:27 Neuron8 kernel: nvidia: module license 'NVIDIA' taints kernel.
 Jan 05 12:50:27 Neuron8 kernel: Disabling lock debugging due to kernel taint
[... 287 lines cut ...]
 core id		: 3
 cpu cores	: 4
 apicid		: 7
 initial apicid	: 7
 fpu		: yes
 fpu_exception	: yes
 cpuid level	: 13
 wp		: yes
 flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx lahf_lm epb tpr_shadow vnmi flexpriority ept vpid xsaveopt dtherm ida arat pln pts
 bugs		:
 bogomips	: 7204.33
 clflush size	: 64
 cache_alignment	: 64
 address sizes	: 46 bits physical, 48 bits virtual
 power management:
ProcVersionSignature: Ubuntu 4.10.0-42.46~16.04.1-generic 4.10.17
RelatedPackageVersions:
 dpkg 1.18.4ubuntu1.3
 apt  1.2.24
Tags: xenial third-party-packages
Uname: Linux 4.10.0-42-generic x86_64
UnreportableReason: This is not an official Ubuntu package. Please remove any third party package and try again.
UpgradeStatus: No upgrade log present (probably fresh install)
_MarkForUpload: True
```

Haveon, 2018-01-26:
It looks like the same file was overwritten, since it failed for both the 4.4 kernel and the 4.10 kernel. Does the patch only work when there is only one kernel installed?

ev-mp, 2018-01-28:
DKMS is is configured to install patches for kernels 4.4 and 4.10 simultaneously. So when it was invoked for 4.13 then it obviously failed.
Kernel 4.4 on the other hand is one of the kernel that we use, therefore we need to understand what went wrong in your case-
- Please verify that you have all the kernel packages installed and active: linux-image, linux-headers, linux-image-extra. It could be the case that when switching to  4.13 the kernel sources/header packages were reconfigured and now point to the incorrect version.
- Additionally, please add the log file(s) for 4.4 build that appeared the original post
> Check /var/lib/dkms/realsense-uvcvideo/4.4.0/build/ for more information.

Haveon, 2018-01-29:
I checked make.log in /var/lib/dkms/realsense-uvcvideo/4.4.0/build found this

```
DKMS make.log for realsense-uvcvideo-4.4.0 for kernel 4.4.0-98-generic (x86_64)
Thu Jan 25 15:43:01 EST 2018
make: Entering directory '/usr/src/linux-headers-4.4.0-98-generic'
make: Makefile: No such file or directory
make: *** No rule to make target 'Makefile'.  Stop.
make: Leaving directory '/usr/src/linux-headers-4.4.0-98-generic'
```
Which prompted me to check /usr/src/linux-headers-4.4.0-98-generic. When I did so I found that all the files were symlinked recursively to themselves (??). I uninstalled linux-image, linux-image-extra and linux-headers using:

```
sudo apt-get --purge remove linux-image-4.4.0-98-generic linux-image-extra-4.4.0-98-generic linux-headers-4.4.0-98-generic
```

After a reboot into the 4.13 kernel, I re-installed the 4.4.0-98 kernel, booted into it saw that the header files were fixed, but my CUDA installation was gone. I reinstalled CUDA 9.1 again, rebooted and saw the symlinks broken again... So I rebooted (again) and magically the headers fixed themselves.

I went through the instructions in https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md#installing-the-packages again. When installing ```realsense-uvcvideo``` I got a similar error asking to check the /var/lib/dkms/realsense-uvcvideo/4.4.0/build again. The make.log:

```
DKMS make.log for realsense-uvcvideo-4.4.0 for kernel 4.13.0-32-generic (x86_64)
Mon Jan 29 17:52:15 EST 2018
make: Entering directory '/usr/src/linux-headers-4.13.0-32-generic'
  AR      /var/lib/dkms/realsense-uvcvideo/4.4.0/build/drivers/media/usb/uvc/built-in.o
  CC [M]  /var/lib/dkms/realsense-uvcvideo/4.4.0/build/drivers/media/usb/uvc/uvc_driver.o
/var/lib/dkms/realsense-uvcvideo/4.4.0/build/drivers/media/usb/uvc/uvc_driver.c: In function ‘uvc_delete’:
/var/lib/dkms/realsense-uvcvideo/4.4.0/build/drivers/media/usb/uvc/uvc_driver.c:1783:34: error: passing argument 1 of ‘media_devnode_is_registered’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  if (media_devnode_is_registered(&dev->mdev.devnode))
                                  ^
In file included from ./include/media/media-device.h:25:0,
                 from /var/lib/dkms/realsense-uvcvideo/4.4.0/build/drivers/media/usb/uvc/uvcvideo.h:14,
                 from /var/lib/dkms/realsense-uvcvideo/4.4.0/build/drivers/media/usb/uvc/uvc_driver.c:28:
./include/media/media-devnode.h:168:19: note: expected ‘struct media_devnode *’ but argument is of type ‘struct media_devnode **’
 static inline int media_devnode_is_registered(struct media_devnode *devnode)
                   ^
cc1: some warnings being treated as errors
scripts/Makefile.build:308: recipe for target '/var/lib/dkms/realsense-uvcvideo/4.4.0/build/drivers/media/usb/uvc/uvc_driver.o' failed
make[1]: *** [/var/lib/dkms/realsense-uvcvideo/4.4.0/build/drivers/media/usb/uvc/uvc_driver.o] Error 1
Makefile:1550: recipe for target '_module_/var/lib/dkms/realsense-uvcvideo/4.4.0/build/drivers/media/usb/uvc' failed
make: *** [_module_/var/lib/dkms/realsense-uvcvideo/4.4.0/build/drivers/media/usb/uvc] Error 2
make: Leaving directory '/usr/src/linux-headers-4.13.0-32-generic'
```

ev-mp, 2018-01-31:
Hi @Haveon , there are references to 4.13 libraries in the log.
I'd suggest to leave DKMS for 4.4 kernel only and remove all other version in the following manner:
> dkms status

Then for each module that is not 4.4.0 do the following:

>sudo dkms remove -m realsense-uvcvideo/4.XX.0 --all

XX for 4.**13**.0 or similar in the above case.
Then apply
>dkms autoinstall

To rebuild all dkms modules for the current kernel and update here.

Haveon, 2018-01-31:
Running ```dkms status```:

```
bbswitch, 0.8, 4.4.0-98-generic, x86_64: installed
nvidia-387, 387.26, 4.4.0-98-generic, x86_64: installed
realsense-uvcvideo, 4.10.0: added
realsense-uvcvideo, 4.4.0, 4.4.0-98-generic, x86_64: installed

```

The only module that lists anything above 4.4 is realsense 4.10

```
sudo dkms remove -m realsense-uvcvideo/4.10.0 --all

------------------------------
Deleting module version: 4.10.0
completely from the DKMS tree.
------------------------------
Done.
```

Then applied ```sudo dkms autoinstall```, running ```dkms status``` again:

```
bbswitch, 0.8, 4.4.0-98-generic, x86_64: installed
nvidia-387, 387.26, 4.4.0-98-generic, x86_64: installed
realsense-uvcvideo, 4.4.0, 4.4.0-98-generic, x86_64: installed
```

Realsense viewer now detects the SR 300, and it seems like my previous code is working as it used to before.

I'm getting a python memory error:
```
*** Error in `python3': free(): invalid pointer: 0x00007f420eeb2cac ***
```
which eventually results in the core being dumped, but this is probably a separate issue.

Thanks for the help!

closed 2018-02-01 by ev-mp