realsenseai/librealsense #1076: Please consider to publish the API reference on Web

UnaNancyOwen, 2018-01-25:
Currently, the API reference needs to be generated using doxygen by users himself. 
But, It is feels very inconvenient.
Please consider to publish the API reference on Web (GitHub Pages, Intel Web Sites, or other).

Thanks,

jlgarcia75, 2018-02-14:
Hello @UnaNancyOwen,

I will send your request to the RealSense team for their consideration.

Regards,
Jesus
Intel Customer Support

closed 2018-02-27 by jlgarcia75

dorodnic, 2018-06-05:
Hi @UnaNancyOwen 
The doxygen output is now published [here](http://intelrealsense.github.io/librealsense/doxygen/annotated.html)

UnaNancyOwen, 2018-06-05:
@dorodnic It's good news! I've been looking forward to it for a long time. 👍 
I will share it with everyone. Thanks,