realsenseai/librealsense #1077: Depth scale is reported to be 0.000000 on D435

tchapi, 2018-01-25:
| Required Info  |   |
|---|---|
| Camera Model | D435 | 
| Firmware Version | 05.08.15.00  | 
| Operating System & Version | Ubuntu 16.04.2 LTS |
| Kernel Version (Linux Only)  | 4.4.0-101-generic  |
| SDK Version | latest master |

Hi,
When I try to retrieve the scale, I get 0.000000 : 

```
auto devices = ctx.query_devices(); // Get a snapshot of currently connected devices
device = devices.front();
auto sensor = device.first<rs2::depth_sensor>();
double scale = sensor.get_depth_scale();
// scale is 0.000000
```

Is this due to the fact that the lib is not D435-ready yet or is this another issue ? Or maybe it is normal with these cameras ?
Thanks a lot !

icarpis, 2018-01-25:
Hello @tchapi, please try to call sensor.get_depth_scale(...) after calling to sensor.open(...)

tchapi, 2018-01-26:
Hi @icarpis and thanks for your answer. I already called `pipe.start()` so calling `sensor.open()` seems to be a no-go : 

    open(...) failed. UVC device is streaming!

This is what I do based on your advice :
```
rs2::pipeline_profile selection = pipe.start();
rs2::device selected_device = selection.get_device();

auto depth_sensor = selected_device.first<rs2::depth_sensor>();
depth_sensor.open(selection.get_stream(RS2_STREAM_DEPTH));
auto scale = depth_sensor.get_depth_scale();
depth_sensor.close();

// ...

pipe.stop();
```

Again, with the SR300 I can retrieve the scale without opening the sensor and it works well — is this a difference with the D435 ?

tchapi, 2018-01-26:
Moreover https://github.com/IntelRealSense/librealsense/blob/master/examples/sensor-control/api_how_to.h#L193 tells me that we don't need to open

icarpis, 2018-01-28:
@tchapi, please run `rs-enumerate-devices -o` and post the output. Also, try to retrieve the depth units using get_option(...) method by the following line: `auto depth_units = sensor.get_option(RS2_OPTION_DEPTH_UNITS);`

tchapi, 2018-01-29:
Indeed, 

    auto depth_units = sensor.get_option(RS2_OPTION_DEPTH_UNITS);

gives `0.001000` for the D435, which looks pretty much correct. And a correct value for the SR300.

Thanks a lot !

closed 2018-01-29 by tchapi