realsenseai/librealsense #1078: Is RS2_STREAM_POSE going to be available on D435 models?

salovision, 2018-01-25:
| Required Info  |   |
|---|---|
| Camera Model | D435 | 
| Firmware Version | 05.08.15.00  | 
| Operating System & Version |  Windows 10  |
| Kernel Version (Linux Only)  |   |
| SDK Version | 2.9.0 (latest from github) |

RS2_STREAM_POSE is defined as a stream type, but I haven't been able to capture any data with that stream.

Is RS2_STREAM_POSE going to be availble on D435?  Does the hardware support it?

zivsha, 2018-01-26:
Hi @salovision,
The Pose stream is meant to represent a stream six degrees of freedom (6DOF) data, which is not the data depth cameras are meant to provide. D435 provides Depth, Infrared and Color images only.
The pose stream is there as future support for the [Intel RealSense Tracking Module T260](https://realsense.intel.com/solutions/#virtual-reality)

salovision, 2018-01-26:
Ok, thanks for clarification!

closed 2018-01-26 by salovision