realsenseai/librealsense #1079: manual Exposure/gain controls

svensht2, 2018-01-25:
| Required Info  |   |
|---|---|
| Camera Model | D415 | 
| Firmware Version |  5.8.14 | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  |  4.13.0-31 |
| SDK Version | 2.8.3 |

Hi, we're in a multi camera setup and we're trying to fix exposure/gain for all cameras.
When I disable auto exposure, I am not able to get the same level of image brightness as it gets with auto mode.
I seem to be able to increase the exposure to where I want it to be(16-33ms), but with gain at 128(max) the image still looks super dark.

zivsha, 2018-01-28:
Hi @svensht2 ,

Can you please upload some snapshots of the RealSense Viewer with the `Controls` list open so we can see what you are getting?
When I disable the Auto Exposure and change the Exposure and Gain manually I get a much brighter image..

svensht2, 2018-01-28:
I’ll send it tomorrow. Are you changing the exposure above 330? anything
above makes it less than 30 FPS.

zivsha, 2018-01-28:
Are we talking about depth or color stream?

svensht2, 2018-01-28:
Sorry for not specifying, I am talking about color stream

zivsha, 2018-02-03:
@svensht2 any updates?

svensht2, 2018-02-03:
@zivsha sorry, didn't take the screenshot, I figured you had issues reproducing the issue since I didn't specify I'm talking about color. Were you not able to reproduce the issue with the specified settings above?

zivsha, 2018-02-04:
Oh sorry I checked this with the depth image. Yes I see now what you are talking about and I'll check what can be done to get the same results w/o AE

svensht2, 2018-03-21:
@zivsha any updates on this?

zivsha, 2018-03-21:
Hi @svenevs , sorry but I couldn't find an answer. 
I'll refer this to @dorodnic with hopes that he will be able to help

svensht2, 2018-05-14:
@zivsha @dorodnic @RealSense-Customer-Engineering  anything new on this?

DillonSkeehan, 2018-05-21:
Required Info |  
-- | --
Camera Model | D415
Firmware Version | 5.9.11.0
Operating System & Version | Ubuntu 16.04
Kernel Version (Linux Only) | 4.13.0-31
SDK Version | 2.11.1

@zivsha @dorodnic @RealSense-Customer-Engineering I am a coworker of @svensht2. Here are the images showing the difference between ae and manual exposure/gain. Manual exposure/gain was set to 330/128.

auto
[image]

manual
[image]

jkosmach, 2018-05-25:
@zivsha @dorodnic @RealSense-Customer-Engineering, Just to add to the above ...  the auto exposure rgb video at 30fps looks great.  when we try to find a corresponding exposure/gain with manual settings at 30fps we can not find it --- the image is just too dark.  In the above, manual settings the gain was set to the maximum and the exposure was set to the maximum to achieve 30fps.   As can be seen, the image is just too dark.  The only way to brighten it up, in order to get anything that resembles the auto exposure case @ 30fps, is to increase the exposure and go well below 30fps.   So to us, something is not right with the exposure/gain parameters that we are allowed to adjust.

jkosmach, 2018-05-30:
@zivsha @dorodnic @RealSense-Customer-Engineering ,   We opened a Zendesk ticket and the engineer there has confirmed there is an issue and they have filed a bug with the fw team.  It looks like the gain setting under manual exposure mode can not be set to it's maximum (which is achieved during auto-exposure).  So this explains why we are seeing unacceptable results for trying to use manual exposure - since we can't set the gain high enough to give us an acceptable picture at 30fps which we see achieved with auto exposure.  
@RealSense-Customer-Engineering, can you verify what is the maximum gain setting  under auto exposure and how it compares to the manual gain maximum of 128?

RealSense-Customer-Engineering, 2018-05-31:
[Realsense Customer Engineering Team Comment]
Hi @jkosmach,

Let's track this on Zendesk directly to get more info although I can't reproduce it on my environment. By the way, what's the Zendesk ticket number?

jkosmach, 2018-05-31:
@RealSense-Customer-Engineering 

The ticket number is #2613, though the subject and initial discussion is around multi-camera sync. I"ve since asked if we should create a separate ticket on it.

With regards to you not being able to reproduce it i'd like to understand this.  So here are the steps to reproduce.  Please confirm this is what you have done or are observing:

1.  start with reasonably good office lighting 
2.  have auto exposure enabled
3. observe the video is at 30fps and brightness level of video is reasonable/usable. Similar to good photo above.
4.  turn off auto exposure (manual exposure)  
5. At this point video should be almost black with default manual exposure and gain settings
6.  Try to adjust exposure and gain settings to get anything close to what you observe with auto exposure.   But make sure video is still at 30fps.   What we find is the best we can do at 30fps is the above dark/unusable video.  The only way to brighten it up is to below the 30fps - which means that manual gain is not capable of being set to what the auto exposure gain is capable of doing.

Is this what you are doing?   I talked with Anders yesterday and he is telling me it's a known issue and that it has something to do with the ISP messing with the ability to adjust the gain.

SolmazJaunt, 2018-06-23:
@jkosmach  @RealSense-Customer-Engineering 
Hi any updates on this issue?
We are experiencing the same problem, and couldnot find any manual setting to match the AE algorithm.

al093, 2018-08-30:
I can still observer this issue in my setup as well. But just to confirm is this solved in any recent release? Or does anyone have a hack to solve it?

SolmazJaunt, 2018-09-04:
@al093 the issue comes from the camera's digital gain, our work around was to introduce a new gain. you can amplify the RGB values to get a better image. 
Your manual and auto setting still would not match, but at least you can get somehow similar images.

VasuAgrawal, 2018-10-05:
@dorodnic I'm using the latest version of the firmware (05.10.03.00) and am also having this issue on my D435. I'm seeing a much brighter image with autoexposure than what I can get if I max out gain with a 33 ms exposure.

With autoexposure: https://drive.google.com/file/d/1PFh-pgOwbgaHT1_G5mqEhdLCMo-qCq5j/view?usp=sharing

Without autoexposure: https://drive.google.com/file/d/1wR2e7YovhFNcIq9AeyGpAn4SJffeWzfa/view?usp=sharing

Is autoexposure doing more than just setting the gain and the exposure time? Is there some way to determine what settings are being used during autoexposure? The realsense viewer doesn't update with the autoexposure settings, and they don't seem to persist when I close the stream in the viewer and then try to query them progammatically.

RealSense-Customer-Engineering, 2018-10-05:
[Realsense Customer Engineering Team Comment]
Please wait for next FW release which should improve the gain range under RGB manual exposure mode.
You can also see #2218 about RGB metadata exposure limitation.

svensht2, 2018-10-05:
Do you have an estimation when that will be available?

VasuAgrawal, 2018-10-08:
@RealSense-Customer-Engineering just bumping @svensht2's question. I'm hoping to be able to use a few RealSense cameras in a low light project I have coming up, but won't be able to do that if the manual gain doesn't work.

VasuAgrawal, 2018-10-29:
@RealSense-Customer-Engineering It seems there was a new firmware release recently (5.10.6). I flashed that firmware onto my RealSense D435, but I have not observed an improvement in the manual gain capacities. I'm seeing the same results as before.

RealSense-Customer-Engineering, 2018-10-29:
[Realsense Customer Engineering Team Comment]
Hi @VasuAgrawal,

Please wait for next firmware release. The solution is not yet integrated in firmware 5.10.6.

RealSense-Customer-Engineering, 2018-12-05:
[Realsense Customer Engineering Team Comment]
A new firmware (5.10.13) is published and should resolve this issue. Please help a try.

closed 2018-12-17 by RealSense-Customer-Engineering