realsenseai/librealsense #1080: Is there a way to authenticate the video stream?

mkhansen-intel, 2018-01-25:
| Required Info  |   |
|---|---|
| Camera Model | R200 / F200 / SR300 / ZR300 / D400 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |
| SDK Version | |

I'm looking to use the RS camera for a secure application in an Intel SGX enclave. I would like to establish a secure authenticated connection to the camera for the video streams. Is there a way to do that using librealsense?

mkhansen-intel, 2018-01-29:
Updated - I'm using Ubuntu 16.04

mkhansen-intel, 2018-02-02:
@zivsha @dorodnic - can anyone answer this?

oordan, 2018-02-02:
There is currently no support for secure applications by RS camera in librealsense.

dorodnic, 2018-02-04:
@mkhansen-intel, please follow-up with @oordan if you have more questions. We currently don't have anything we can publish on this subject.

closed 2018-02-04 by dorodnic

mkhansen-intel, 2018-02-05:
@dorodnic - thanks I'll do that.