realsenseai/librealsense #1081: Python crashes while calculating pointcloud

Panepo, 2018-01-26:
| Required Info  |   |
|---|---|
| Camera Model |  D400 | 
| Firmware Version | 05.08.15.00  | 
| Operating System & Version | Win10 version 1709 (OS Build 16299)  |
| Kernel Version (Linux Only)  |   |
| SDK Version | 2.9.0 |

Hi there,
I am trying to determine point cloud on python wrapper and find a problem. The python environment ALWAYS crashes if the resolution of the color and depth is not the same. It seems crash at the following position:

```
    # Tell pointcloud object to map to this color frame
    pc.map_to(color)

    # Generate the pointcloud and texture mappings
    points = pc.calculate(depth)
```

And there is a message pops on windows event viewer:

> Faulting application name: python.exe, version: 3.6.3150.1013, time stamp: 0x59e338b7
> Faulting module name: ucrtbase.dll, version: 10.0.16299.125, time stamp: 0x70f70cc4
> Exception code: 0xc0000409
> Fault offset: 0x000000000006b70e
> Faulting process id: 0x381c
> Faulting application start time: 0x01d3967a103204a0
> Faulting application path: C:\ProgramData\Anaconda3\python.exe
> Faulting module path: C:\WINDOWS\System32\ucrtbase.dll
> Report Id: af310638-a0c9-4112-a5ed-bc9c05ec5a6e

zivsha, 2018-01-28:
Hi @Panepo ,
I can't seem to reproduce this. Which camera are you using?
I've added the following lines to `export_ply_example.py` and it doesn't occur:
```py
cfg.enable_stream(rs.stream.depth, -1, 640, 480, rs.format.z16, 30)
cfg.enable_stream(rs.stream.color, -1, 1920, 1080, rs.format.rgb8, 30)
```
Which streams were you using to get `depth` and `color`?

Panepo, 2018-01-29:
Hi @zivsha 
I have turned align on, maybe it is the problem?
I've modified `export_ply_example.py`to [this](https://gist.github.com/Panepo/65da57cea20fa68b560f06918a9aa159) to fit my camera config, and this code crashes every time.

zivsha, 2018-02-02:
Hi @Panepo ,
I get a crash as well when trying your code, the reason is that you are mapping the pointcloud texture **before** you calculate it. just reverse the two lines and it should work:
```py
    points = pc.calculate(depth_frame) #First clculate
    pc.map_to(color_frame) #Then map texture
```

zivsha, 2018-02-02:
I've marked as `bug` since it shouldn't crash like that.

zivsha, 2018-03-05:
Does not reproduce with PR #1278 . Closing as fixed. @Panepo if you still see the issue please let us know or open a new issue

closed 2018-03-05 by zivsha