realsenseai/librealsense #1082: linking to rs2_set_devices_changed_callback fails

SirDifferential, 2018-01-26:
| Required Info  |   |
|---|---|
| Camera Model | D400 | 
| Firmware Version | N/A  | 
| Operating System & Version | Debian 9  |
| Kernel Version (Linux Only)  | 4.9.65 with the patches applied  |
| SDK Version | 2.9.1.0, built with `-march=native` |

How does one use the device changed callback from a C application? I get a linker error when trying to use the callback registration functions:

```
#include <stdio.h>
#include <stdint.h>
#include <librealsense2/rs.h>

uint8_t check_error(rs2_error* e)
{   
    if (e)
    {   
        printf("rs_error was raised when calling %s(%s):\n", rs2_get_failed_function(e), rs2_get_failed_args(e));
        printf("    %s\n", rs2_get_error_message(e));
        return 1;
    }
    return 0;
}

void dev_changed(rs2_device_list* removed_devs, rs2_device_list* added_devs, void* ptr)
{

}

int main()
{
    rs2_error* e = NULL;
    rs2_context* ctx = rs2_create_context(RS2_API_VERSION, &e);
    if (check_error(e) != 0)
        return 1;

    rs2_set_devices_changed_callback(ctx, dev_changed, &e);
    if (check_error(e) != 0)
        return 1;

    return 0;
}
```

```
gekko@delidebi:~/setrsadvanced/minimal$ gcc main.c -I../../librealsense/include/ -L../../librealsense/build -lrealsense2
/tmp/cczpNNjU.o: In function `main':
main.c:(.text+0xdb): undefined reference to `rs2_set_devices_changed_callback'
collect2: error: ld returned 1 exit status
```

Looking at the exported symbols in the library it looks as if this function gets C++ -style function name mangling?

```
gekko@delidebi:~/librealsense/build$ nm -g librealsense2.so | grep -i 'rs2_set_devices_changed_callback'
0000000000ac3ea6 T rs2_set_devices_changed_callback_cpp
0000000000ac379d T _Z32rs2_set_devices_changed_callbackPK11rs2_contextPFvP15rs2_device_listS3_PvES4_PP9rs2_error
```

zivsha, 2018-01-28:
Hi @SirDifferential ,

Thanks for reporting this. 
This is indeed a bug and I am able to reproduce it.
The root of this is that `rs2_set_devices_changed_callback` is defined as:
https://github.com/IntelRealSense/librealsense/blob/2c88be0a7747f6d4b082ab61cbe24834518d6e37/include/librealsense2/h/rs_context.h#L47
but implemented as:
https://github.com/IntelRealSense/librealsense/blob/2c88be0a7747f6d4b082ab61cbe24834518d6e37/src/rs.cpp#L658

(`const rs2_context*` is const and `void* user` is missing)

I will open a PR to fix it, but if you are compiling the library yourself you can apply this patch for now:
```diff
diff --git a/include/librealsense2/h/rs_context.h b/include/librealsense2/h/rs_context.h
index 9a367df9..e24648a3 100644
--- a/include/librealsense2/h/rs_context.h
+++ b/include/librealsense2/h/rs_context.h
@@ -44,7 +44,7 @@ void rs2_set_devices_changed_callback_cpp(rs2_context* context, rs2_devices_chan
 * \param[in] callback function pointer to register as per-notifications callback
 * \param[out] error  if non-null, receives any error that occurs during this call, otherwise, errors are ignored
 */
-void rs2_set_devices_changed_callback(rs2_context* context, rs2_devices_changed_callback_ptr callback, rs2_error** error);
+void rs2_set_devices_changed_callback(const rs2_context* context, rs2_devices_changed_callback_ptr callback, void* user, rs2_error** error);
 
 /**
  * Create a new device and add it to the context
```

SirDifferential, 2018-01-28:
Good to know. I'll merge your changes to our fork.

zivsha, 2018-02-02:
@shiritbrook , did the change fix the problem for you? can we close this issue?

SirDifferential, 2018-02-02:
Yes, I can confirm this works.

closed 2018-02-02 by SirDifferential